FlyGoat/RyzenAdj #3: Read configurations before setting them

PlutoniumHeart, 2019-02-16:
Hi,

I am wondering if it's possible to read the configurations from SMU instead of blindly setting them. I tried using the ID=0x43 (GetSustainedPowerAndThmLimit), but I can't make sense of the number I got.

Thanks!

FlyGoat, 2019-02-16:
It seems like we can't get the current value by this method. But it is possible to get from pwrSMU. Check kernel driver of AMD uProf for more information.

PlutoniumHeart, 2019-02-16:
It's the first time I am getting this low level of programming, can you give me some more hints about pwrSMU and the kernel driver of AMD uProf? I think uProf is not open source itn't it?

Thanks!

FlyGoat, 2019-02-18:
See bin/ of uProf Linux version.
There is a archive of kernel driver source code. According to GPL, AMD must open-source it.

storm1er, 2019-03-04:
On the same issue, would be nice to be able to get those informations using `-i` arg =) 
Is this going to be implemented using this?

FlyGoat, 2019-03-04:
> On the same issue, would be nice to be able to get those informations using `-i` arg =)
> Is this going to be implemented using this?

That's my plan.
However probably this function might never be available on Windows since we can't get physical memory access on windows.

FlyGoat, 2019-03-21:
Implemented in ryzen-controller.

closed 2019-03-21 by FlyGoat