FlyGoat/RyzenAdj #6: Stop closing after applying settings

storm1er, 2019-02-25:
Hi guys,

It would be nice if the adj window could have standard features:
- Minimize button
- Close Button
- Apply without closing (or add a tick "Closing on applying settings")

FlyGoat, 2019-02-26:
Thanks for your suggestions.
Just forwarded them to the GUI develper.

storm1er, 2019-02-26:
Actually working on a advanced electron app if you're interested. 
Will be open source btw.

storm1er, 2019-02-27:
@FlyGoat 
I would like to implement your compiled ryzenadj.exe into my project.
Obviously, this repo will keep the development and source code, my project will only contain the compiled ryzenadj.exe and the needed dlls.

See https://gitlab.com/le.storm1er/ryzen-controller.

FlyGoat, 2019-03-03:
Thanks for your works.
Will promote "ryzen-controller" in readme.

closed 2019-03-03 by FlyGoat