FlyGoat/RyzenAdj #7: 0.3 relese GUI Can't find ryzenadj.exe

RUANRUI1995, 2019-03-02:


RUANRUI1995, 2019-03-02:
it says "Can't find ryzenadj.exe in current folder" when i click the Appy button,but this file exists in current folder

FlyGoat, 2019-03-02:
Forwarded to GUI author.
Thanks for reporting.

kingslayer4, 2019-03-02:
The UI is way better than the last one.
But apply option results in an error - "Can't find ryzenadj.exe".

Kennethai, 2019-03-02:
Same issue here, "Can't find ryzenadj.exe". And when attempting to open ryzenadj.exe, as soon as the cmd window opens it closes again. Running as admin does nothing in either case.

JustSkillUA, 2019-03-02:
The error was caused by my inattention. Fix coming later today, when FlyGoat comes online.

FlyGoat, 2019-03-03:
Uploaded.
Sorry for inconvenience.

closed 2019-03-03 by FlyGoat

MKD77, 2019-03-03:
Hi, I really appreciate your work. It is really great to change the TDP of this Ryzen based laptops to gain more performance. But I have something to tell (suggestion) ,that is can you build the ultimate Ryzen mobile apus tool kit. I mean where we can change or set CPU frequency also GPU frequency, Notebook fan speed control . If it is possible we will be highly beneficial .

FlyGoat, 2019-03-03:
@MKD77
Thanks for your suggestions.
CPU frequency is almost done, GPU frequency and infinite fabric bus speed feature is merged in a Pull Request and will release later.

Fan speed is a little bit difficult for me since different laptop have different EC interface.