FlyGoat/RyzenAdj #12: Which CPUs are supported

gsedej, 2019-03-12:
Which Ryzen CPUs are supported?

I tried on my R5 1600x + Gigabyte AB350m gaming 3 on linux. Regardless of all the parameters, I didn't noticed any difference. I was reading temperature and power usage (`rapl-read-ryzen`), under cpu load, but didn't see difference 

`sudo ./ryzenadj --stapm-limit=45000 --fast-limit=45000 --slow-limit=45000 --tctl-temp=80`
or
`sudo ./ryzenadj --stapm-limit=100000 --fast-limit=100000 --slow-limit=100000 --tctl-temp=100`
gave same reading.

My friend using HP laptop with 2700u (also on linux) did notice a bi improvement.

Also, Is it possible to read those values? To confirm if the values were set correctly.

GoelBiju, 2019-03-12:
@gsedej you could view some of the values related to STAPM through AMD uProf (https://developer.amd.com/amd-uprof/). We cannot, presently, read the thermal limits which is enforced by the System Management Unit but you might be able to see it in action when testing some intensive applications. 

All of the Ryzen Mobile range and the Ryzen desktop should be supported. I believe the desktop processors are unlocked so you might not see much difference changing the STAPM alone as it would have already been set appropriate to the system.

gsedej, 2019-03-12:
Thanks for quick reply. I was not aware of such apps from AMD and that they work under linux.

I would be helpful if put such info would be on main project page (cpu support and amd-uprof)

GoelBiju, 2019-03-14:
@gsedej that would be a useful addition to the README. Thank you.

closed 2019-05-05 by FlyGoat