FlyGoat/RyzenAdj #16: TDP settings keep restoring to default even though the app is running in background

n3onis, 2019-03-21:
Is this a known issue? I've had no success changing TDP limits. They do change at first, but then they reset every time.

n3onis, 2019-03-21:
Found a [comment on reddit](https://www.reddit.com/r/Amd/comments/anbl95/ryzen_master_for_mobile_ryzen_modify/efv3wwr?utm_source=share&utm_medium=web2x) stating the same. I wonder what you need to fix this?

n3onis, 2019-03-21:
Just found your [comment on reddit](https://www.reddit.com/r/Amd/comments/anbl95/ryzen_master_for_mobile_ryzen_modify/efv3wwr?utm_source=share&utm_medium=web2x). I can extract my DSDT and send it to you, if it helps.

FlyGoat, 2019-03-22:
In fact I am still not sure about the reason. But you can try to set stapm time into a very big value. Some users reported it may help.

FlyGoat, 2019-03-22:
Remove ALIB calls in dsdt works on some machines but not all.

speedracerlo, 2019-04-11:
I have the same problem here.  I've used v0.3 and it works at first, but sometime later it reverts back to original.  How do I remove ALIB calls?

n3onis, 2019-04-11:
> 
> 
> I have the same problem here. I've used v0.3 and it works at first, but sometime later it reverts back to original. How do I remove ALIB calls?

The best solution I've found for now is using the Ryzen Controller GUI alternative linked on this repo and setting the re-apply timer to a few seconds.

closed 2019-09-23 by FlyGoat