FlyGoat/RyzenAdj #17: not working on Acer Nitro 5

Benman2785, 2019-04-07:
hi, it doesnt work on my Acer Nitro 5 with Ryzen 5 2500U
no difference between 12W or 35W

FlyGoat, 2019-04-10:
Hi, a lot of dudes have verified it works perfectly on your model.
Could you please use AMD uprof to check if the change valid?

Benman2785, 2019-04-10:
[image]
it changes the limits but CPU still limits itself and i dont know why!

Benman2785, 2019-04-10:
[image]
highest performance i have running the APU on stock - without changing any value

FlyGoat, 2019-04-11:
Did you perform any BIOS update? 
So it should be something wrong with SMU.

Benman2785, 2019-04-11:
i use 1.13
what is the SMU i need to use?

Benman2785, 2019-04-15:
i use SMU v30.80

closed 2019-06-04 by FlyGoat