FlyGoat/RyzenAdj #18: Doesn't work on Ryzen 5 3550H

leonmaxx, 2019-04-22:
```
$ sudo ./ryzenadj -i
Not a Ryzen NB SMU, BIOS Interface Ver: 0xf
Unable to init ryzenadj, check permission
```

Is it safe to add check for 0x0f to [api.c:36](https://github.com/FlyGoat/RyzenAdj/blob/8bf44ed6e0d53e01c774f09734e53a731b9f2287/lib/api.c#L36)

GoelBiju, 2019-04-23:
@leonmaxx , @FlyGoat may know the answer to this but it might worth trying out to see if it works or not. I am not sure if the SMU interface is the same on the new Ryzen 3rd generation mobile processors but if they are using the same AGESA protocol then it might work.

FlyGoat, 2019-04-23:
It should be safe. If anything went wrong, a reboot can simply save you.

leonmaxx, 2019-04-23:
I tied and it seems to work:
```
$ sudo ./ryzenadj --tctl-temp=80
Sucessfully set tctl_temp to 50
```

FlyGoat, 2019-04-24:
@leonmaxx 
Can AMD uProf read out the change?

leonmaxx, 2019-04-24:
@FlyGoat  I'll try later today and report.

leonmaxx, 2019-04-24:
Unfortunately I cannot install AMD uProf it's kernel driver fails to build on Linux 5.0.x.

dublollo, 2019-06-01:
hi, i have the same issue with windows how can i solve?

FlyGoat, 2019-06-04:
@dublollo Please try the latest version.

dennisgyor, 2019-10-19:
This is still not fixed. Im not sure why this issue was closed, it seems unclear as to what the fix was. I have the latest version and I am getting a similar error:

Not a Ryzen NB SMU, BIOS Interface Ver: 0x1Unable to init ryzenadj, check permission

FlyGoat, 2019-10-20:
> This is still not fixed. Im not sure why this issue was closed, it seems unclear as to what the fix was. I have the latest version and I am getting a similar error:
> 
> Not a Ryzen NB SMU, BIOS Interface Ver: 0x1Unable to init ryzenadj, check permission

Hi @dennisgyor, as far as I know, BIOS Interface Ver: 0x1 is not supported by ryzenadj, the minimal interface version requirement is 0x5, could you please provide some detail information about your machine? I haven't seen such an early version before.

FlyGoat, 2019-10-20:
The issue was closed because  I removed the maximum version check but let minimal version check remained. Your interface version is earlier than our minimal version.

CxZMoE, 2020-07-01:
> ```
> $ sudo ./ryzenadj -i
> Not a Ryzen NB SMU, BIOS Interface Ver: 0xf
> Unable to init ryzenadj, check permission
> ```
> 
> Is it safe to add check for 0x0f to [api.c:36](https://github.com/FlyGoat/RyzenAdj/blob/8bf44ed6e0d53e01c774f09734e53a731b9f2287/lib/api.c#L36) ?

In my laptop this tool works.
Maybe you need to enable kernal parameter: `amdgpu.ppfeaturemask=0xffffffff`,according to [archwik#overclockingi](https://wiki.archlinux.org/index.php/AMDGPU#Overclocking)

Or should be the problem of kernel :smile: 
my kernel version is `https://wiki.archlinux.org/index.php/AMDGPU#Overclocking`

leonmaxx, 2020-07-01:
@CxZMoE This was fixed since Oct 2019, and it works for you because it's already fixed!

jpvmet0, 2020-11-21:
Hi i have the same problem with ryzen 7 3700u from hp 15s-eq0262ng, i hace the bios/uefi locked, and the cpu was locked to use 400mgz AND the Max Is 2300, i want to use ryzenadj to.change this, i instaló Ubuntu 20.03 AND compile ryzenadj i can put the ryzen -h un the Shell but i cant change any value, the message of 
Not a Ryzen NB SMU, BIOS Interface Ver: 0xf
Unable to init ryzenadj, check permission starts

CxZMoE, 2020-11-28:
> Hi i have the same problem with ryzen 7 3700u from hp 15s-eq0262ng, i hace the bios/uefi locked, and the cpu was locked to use 400mgz AND the Max Is 2300, i want to use ryzenadj to.change this, i [...]

Did you check your permissions?

jpvmet0, 2020-11-28:
Hi !!

Where i can check the permision? I excecuted in sudo mode.

There is more? If the permision are un the BIOS its imposible because HP
locked the advanced mode.

If You can help i'm going to thanks a lot

CxZMoE, 2020-11-29:
> Hi !! Where i can check the permision? I excecuted in sudo mode. The message is : Not a Ryzen NB SMU, BIOS Interface Ver: 0x1unable to init ryzenadj, check permision There is more? If the permision are in the BIOS its imposible because HP locked the advanced mode. If You can help i'm going to thanks a lot El sáb., 28 de noviembre de 2020 01:32, cxzmoe <notifications@github.com> escribió:
> […](#)

You can try switching to root user and then execute the command. If that doesn't make a difference,maybe you should make a compatibility issue to the auther/

FlyGoat, 2020-11-29:
There are some reports that secure boot denies PCI CFG space access.
You can try to switch it off in BIOS.

FlyGoat, 2020-11-29:
Oops it's not a permission issue. There is a new interface version?

jpvmet0, 2020-11-29:
I supose because , i tried touching everything in the BIOS but i can't solve the issue. I tried to use ryzenadj only to move the temperature control because the Genius of HP locked un 50 C and i only reachs 0.399Ghz is a terrible experiencia. In the BIOS the advanced options are locked and i tried everything that you can find in Google, ctrl F1, press A, put ole BIOS versión, but nothing works. The especific of all the things that i tried 

https://h30467.www3.hp.com/t5/Software-para-notebooks/HP-15s-eq0262ng-desbloquear-bios-o-procesador/m-p/1123038#M60027

jpvmet0, 2020-11-29:
> Hay algunos informes de que el arranque seguro niega el acceso al espacio PCI CFG.
> Puede intentar apagarlo en BIOS.

Hi i can lunch ryzenadj with turno of the save init. The thing is the i can use ryzenadj but nothing change un the cpu
[image]

All the cpus are locked un 399.272.

Any idea ?

CxZMoE, 2020-11-30:
> > Hay algunos informes de que el arranque seguro niega el acceso al espacio PCI CFG.
> > Puede intentar apagarlo en BIOS.
> 
> Hi i can lunch ryzenadj with turno of the save init. The thing is the i can use ryzenadj but nothing change un the cpu
> [image]
> 
> All the cpus are locked un 399.272.
> 
> Any idea ?

check your battery connection,loose battery connection could cause this.

jpvmet0, 2020-11-30:
I check the batery conection when i take the photo i was with the Charger plugin.

Falcosc, 2021-02-23:
@jpvmet0 so basically you try to fix the broken HP System with ryzenAdj
Before you try to achieve this you need to check first if the SMU does actually have an 50°C limit applied

Use a tool to monitor the PTable, I can tell you about 3-4 windows tools to get the table but I don't know how you can do it on your system.
If you need to do it manually you, these are the adresses. You need to send things to the SMU and read things directly from memory
```
Rsmu.SMU_ADDR_MSG = 0x03B10A20;
Rsmu.SMU_ADDR_RSP = 0x03B10A80;
Rsmu.SMU_ADDR_ARG = 0x03B10A88;

Rsmu.SMU_MSG_GetTableVersion = 0x6;
Rsmu.SMU_MSG_TransferTableToDram = 0x65;
Rsmu.SMU_MSG_GetDramBaseAddress = 0x66;
```

And here are the offset where you find the needed readings: https://github.com/sbski/Renoir-Mobile-Tuning/blob/master/renoir_tuning_utility/renoir_tuning_utility/SystemMonitor.cs

You need to check the following values
```
 "STAPM LIMIT",    0x0
 "STAPM VALUE",    0x4
 "PPT LIMIT FAST", 0x8
 "PPT VALUE FAST", 0xC
 "PPT LIMIT SLOW", 0x10
 "PPT VALUE SLOW", 0x14
 "PPT LIMIT APU",  0x18
 "PPT VALUE APU",  0x1C
 "TDC LIMIT VDD",  0x20
 "TDC VALUE VDD",  0x24
 "EDC LIMIT VDD",  0x30
 "EDC VALUE VDD",  0x34
 "THM LIMIT CORE", 0x40
 "THM VALUE CORE", 0x44
 "STT LIMIT APU",  0x58
 "STT VALUE APU",  0x5C
 "PROCHOT", 0x870 or 0x88C
```

If one of the `VALUE` readings are equal to the according `LIMIT` reading then you have a chance to remove the limit. 
If none of these values report a limit hit, then your system it throttled by something else and could not be fixed.

The Next step would be to change the causing limit with ryzenAdj and check PTable if it got applied.

I wish you luck!

mirh, 2021-02-24:
Don't you have to switch thermal profile in HP command center to have settings stick?

Speaking of which, do you know if those couldn't just correspond to the [ACPI Platform Profile](https://lore.kernel.org/patchwork/patch/1382847/)?

FlyGoat, 2021-02-24:
> Don't you have to switch thermal profile in HP command center to have settings stick?
> 
> Speaking of which, do you know if those couldn't just correspond to the [ACPI Platform Profile](https://lore.kernel.org/patchwork/patch/1382847/)?

At least the backend of Lenovo platform profile DYTC will use SMU to adjust power limits.

I'm the author of ideapad platform-profile as well :-)

Falcosc, 2021-02-24:
> At least the backend of Lenovo platform profile DYTC will use SMU to adjust power limits.
>
> I'm the author of ideapad platform-profile as well :-)

If you got the undocumented knowledge before RyzenAdj in the Lenovo Project, then you are a cheater :P 
I did notice message Ids popping up in the SMU mailbox monitoring during power source change as well on my Lenovo T14s but didn't know what these are doing.

Is it true that there is currently no linux tool available for structured readout of the power table? I found only tools with basic support for temps and actual power readings, but nothing with all limit values.
Closest tool I found was https://gitlab.com/leogx9r/ryzen_smu, but it does dump the whole thing as blob.

FlyGoat, 2021-02-24:
> If you got the undocumented knowledge before RyzenAdj in the Lenovo Project, then you are a cheater :P

Oh I meant Linux Kernel patch of platform profile support. That's also based on reverse engineering of Windows driver and ACPI tables.

jpvmet0, 2021-02-24:
Thanks a Lot for the advice im going to try that because i still have the
problem.

The cpu never works up of 50.

In the bios there aré no info about this.

Falcosc, 2021-02-24:
Meanwhile, I found a more easy way for Linux users 

you can install [ryzen_smu](https://gitlab.com/leogx9r/ryzen_smu) and use one of his scripts [scripts/monitor_cpu.py](https://gitlab.com/leogx9r/ryzen_smu/-/blob/master/scripts/monitor_cpu.py), but you may need to change it a bit to find the values referenced in our descriptions.

closed 2021-04-01 by FlyGoat

jpvmet0, 2021-04-01:
Tha.nks but the locker in the cpu Is imposible to chance. I tried all

commit 3b7667af (2021-03-24): Add table entries for Raven/Picasso and fix table commands
Resolves #18
commit 2a8276ed (2021-03-24): Add table entries for Raven/Picasso and fix table commands
Resolves #18
commit 88c8a9d9 (2021-03-25): Add table entries for Raven/Picasso and fix table commands
Resolves #18
commit cb527703 (2021-04-01): Add table entries for Raven/Picasso and fix table commands
Resolves #18