FlyGoat/RyzenAdj #20: Not working on ryzen 5 2500u on ubuntu

Yrahate7, 2019-05-05:
I ran the program as root user,but it shows the following
[image]

FlyGoat, 2019-05-05:
Some users reported Ubuntu declined PCI config space RW too, I have no idea about that.
It works fine on Debian and ArchLinux.
I'm not sure if we can unlock it on Ubuntu.

GoelBiju, 2019-06-07:
@FlyGoat I have tested it on Ubuntu 18.04 LTS. It works fine as long as you use it as a sudo user.

chaos-dremel, 2020-01-02:
@FlyGoat @Yrahate7 I had same issue. But when I disabled secure boot (in BIOS) the problem "magically" disappeared.

Yrahate7, 2020-01-03:
@FlyGoat @chaos-dremel , Can confirm. Disabling Secure boot in BIOS fixes the issue

closed 2020-01-03 by Yrahate7