FlyGoat/RyzenAdj #24: doesn't work anymore with Ryzen 5 2500u new bios update hp F.31

shakumaru, 2019-05-25:
Not a Ryzen NB SMU, BIOS Interface Ver: 0x7
Unable to init ryzenadj, check permission

razorbackdb, 2019-05-29:
Can confirm, same happened to me too, both in windows and linux. Triend to downgrade the bios but no use, hp seems to have blocked the downgrade at least on my case. Laptop is HP Pavilion - 15-cw0005la.

FlyGoat, 2019-06-02:
Seems like AMD added a lot of new Interface Version.
I'm going to disable the check.

FlyGoat, 2019-06-03:
Latest code had fixed the issue.
I'm going to make a release later.

shakumaru, 2019-06-03:
should it works with the last version (1.12.3)? the issue still there

razorbackdb, 2019-06-04:
@FlyGoat worked for me on windows, thank you!, haven't tried linux yet, @shakumaru try this build
[x64-Release.zip](https://github.com/FlyGoat/RyzenAdj/files/3250006/x64-Release.zip)

FlyGoat, 2019-06-04:
The latest release fixed the issue,

shakumaru, 2019-06-04:
thanks @razorbackdb the problem was ryzenadj.exe i dont know why

closed 2019-06-04 by shakumaru