FlyGoat/RyzenAdj #25: [Docs] Looking for option's description

storm1er, 2019-06-02:
Hi guys,

I'm the developer of Ryzen Controller,
We are looking for options description, see [here](https://gitlab.com/le.storm1er/ryzen-controller/issues/17).

We already implemented some, see [here](https://gitlab.com/le.storm1er/ryzen-controller/blob/master/js/options_description.json). But they probably don't have the perfect meaning.

Could you help us?

I also would like to get some of RyzenAdj developers into RyzenController members, so you'll be able to interact with the repo. (as merge request and issue approvers/reviewers).

Thanks.

FlyGoat, 2019-06-02:
For me, most information was based on guess.
Don't know if it's possible to get some official information.

storm1er, 2019-06-03:
I think that "official" description isn't necessary,
We just need the basic description to tell the user what to expect from changing the setting described.

FlyGoat, 2019-06-05:
https://github.com/FlyGoat/RyzenAdj/wiki/Options
Created a Wiki page but no clues about what to write.

Road-Drum, 2019-09-19:
some of the infos are vague. there's VRM current limit and at the same time there are also VRM SoC limit. not to mention PSI0 which is ayylmao to me. fclk, highly likely is the fabric clock as seen in zen2 board options.

can't seem to find any leaked documentation for this stuff. some stuff can be guessed (stapm limit, gfxclock) while some can't

commit 9a5d40d1 (2021-02-24): add documentation for all power, time and temperature options based on Renoir
add FAQ
resolves #25
resolves #41
resolves #42
resolves #48
resolves #49
resolves #50
resolves #53
resolves #58
resolves #72
closed 2021-02-24 by FlyGoat