FlyGoat/RyzenAdj #30: List of experience values for devices?

Trollwut, 2019-06-08:
Hey folks!

Very nice software and I like playing around with it. I've got an E585, which has been a throttled TDP from factory, god knows why.

How about a list of devices and values people here have tested that work in normal conditions?

Like that we have a table for models and users can add their values, so when others want to tinker around they have some values where they can orient themselves at.

FlyGoat, 2019-06-09:
Sorry but read out values is still unimplemented due to technical reason.
Reverse Engineering of uProf shown some ways however still a little bit hard to implement.

FlyGoat, 2019-06-10:
Users are welcomed to apply their experience values to the wiki.

Trollwut, 2019-06-10:
Would maybe creating a template table help users to see that there is something to contribute to?

PR #113 (2021-03-10): Document which Processors are supported and which commands do work
New Support Matrix in https://github.com/FlyGoat/RyzenAdj/wiki/Supported-Models

Resolves #28
Resolves #30
Resolves #33
Resolves #54
Resolves #80
commit 04aa9560 (2021-03-10): Document which Processors are supported
Resolves #28
Resolves #30
Resolves #33
Resolves #54
Resolves #80
closed 2021-03-10 by FlyGoat