FlyGoat/RyzenAdj #31: Some way to output current values

Trollwut, 2019-06-08:
Hi folks,

I'd like to have a way to output the current values that can be manupulated by RyzenAdj.

I tried the `-i` tag, but I can't get any information on Manjaro (newest updated, kernel 5.1.4):

    $ ./ryzenadj -i
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config
    Unable to init ryzenadj, check permission

And with root rights:

    $ sudo ./ryzenadj -i
    
(no output at all)

Any way to troubleshoot this or is this intended behaviour?

Got an E585 with a Ryzen 2500U

FlyGoat, 2019-06-09:
Sorry but read out values is still unimplemented due to technical reason.
Reverse Engineering of uProf shown some ways however still a little bit hard to implement.

Trollwut, 2019-06-09:
Ok, do understand this. Bummer. :D

So there also won't be a simple way to go back to default values after playing around?

FlyGoat, 2019-06-10:
Yes, Unless you reboot it.

closed 2019-06-10 by FlyGoat