FlyGoat/RyzenAdj #32: Is ryzen desktop processors supported?

N46AN, 2019-07-23:
In #12 one of the contributors states that 
> All of the Ryzen Mobile range and the Ryzen desktop should be supported. I believe the desktop processors are unlocked so you might

I tried running this tool with Ryzen 5 2600x on a B450 MB but got:

> > ryzenadj.exe --tctl-temp=60
> Not a Ryzen NB SMU, BIOS Interface Ver: 0x0
> >

OS is Windows 10.

Thanks!

FlyGoat, 2019-07-25:
BIOS Interface 0x0 means CPU have totally no respond to service requests.
I've tested RyzenADJ on R5 2200G with AGESA RavenPi 1.1.0.
But I'm not sure if recent SMU/BIOS update (ComboPi) had changed that interface.

bobafetthotmail, 2019-10-13:
I've ran this on a Ryzen 3 PRO 2200GE (that I bought from Aliexpress, I'm trying to increase its power limits) on a AB350M Pro4 with BIOS version 5.90 https://www.asrock.com/mb/AMD/AB350M%20Pro4/index.asp#BIOS on Linux.

It does not seem to error out, but I have no other way to tell if this actually does anything.

```
sudo ./ryzenadj --stapm-limit=65000 --fast-limit=65000 --slow-limit=65000 --tctl-temp=90
[sudo] password for root: 
Sucessfully set stapm_limit to fde8
Sucessfully set fast_limit to fde8
Sucessfully set slow_limit to fde8
Sucessfully set tctl_temp to 5a
```

OneB1t, 2020-01-09:
temperature is working for me on 200GE but rest like GFX clock im only able break it but cant set it to normal value

OneB1t, 2020-01-11:
Ok now i know how to find if CRavenCPU::IsOverClockable this is MSR register 0xC0010292 first bit 0x10000000 represents if CPU can be overclocked (inside code they are checking that via & 0x1F8000 command)
1 means overclock disabled
0 means overclock enabled

[image]

closed 2020-05-18 by FlyGoat