FlyGoat/RyzenAdj #33: Mobile Ryzen 3000 series

Dragomir-Ivanov, 2019-07-25:
Will this work on newest Zen+ laptops, like T495?

Ryzener, 2019-11-14:
Setting a Maximum SoC Clock Frequency on my Ryzen 5 3500U doesn't seem to work on my Ryzen 5 3500U, which is the same CPU installed on Thinkpad T495. See https://github.com/FlyGoat/RyzenAdj/issues/38.

I'm happy to test other settings for you, if you can tell me what and how to verify it.

leoncryp, 2020-10-15:
Ryzener regarding #38 
to save power the lowest i can get is 400 mhz on all cores 3500u, single core will be higher depending on usage.
sudo ~/RyzenAdj/build/ryzenadj --fast-limit=5000 --slow-limit=5000 --stapm-limit=5000

5000 mA is very low, youtube will not run etc.... you can test other settings eg 10000 mA etc.....

i find its best to just play around with the Amps, the system will take care of everything else automatically 
I like to run mine at 23000 anything above it will start to drain my battery (if i run 100% cpu) as I only have a 45watt psu.

PR #113 (2021-03-10): Document which Processors are supported and which commands do work
New Support Matrix in https://github.com/FlyGoat/RyzenAdj/wiki/Supported-Models

Resolves #28
Resolves #30
Resolves #33
Resolves #54
Resolves #80
commit 04aa9560 (2021-03-10): Document which Processors are supported
Resolves #28
Resolves #30
Resolves #33
Resolves #54
Resolves #80
closed 2021-03-10 by FlyGoat