FlyGoat/RyzenAdj #35: Recently Failed to open on Ryzen 2500u

Nala16, 2019-09-17:
So yesteday i can run Ryzenzontroller just fine, but today i can't even open it.
Here is the Log report

[SquirrelSetup.log](https://github.com/FlyGoat/RyzenAdj/files/3619951/SquirrelSetup.log)
The same goes with Ryzenadj, which is weird.

For information i am using Asus f570zd , Ryzen 5 2500 U, nvidia 1050. I am using Ryzencontroller version 1.15.1

FlyGoat, 2019-09-18:
Sorry, I'm not pretty familiar with Node, and I can't find any RyzenAdj relevant information in your log. Could you please collect the error given by RyzenAdj?
Thanks.

Nala16, 2019-09-21:
Thanks for your reply, nevermind i sorted it out. After i restarted it somehow i can run Ryzencontroller just fine.
I think it has something to do with the Administrator account.

Keep up the good work guys!

closed 2019-09-23 by FlyGoat