FlyGoat/RyzenAdj #38: SoC Clock Frequency not working on Ryzen 5 3500U

Ryzener, 2019-11-14:
I have a Ryzen 5 3500U.
According to the specs, its base frequency is 2.1GHz and it can go up to 3.7GHz.
<sub>At the minimum system load `/proc/cpuinfo` shows the `cpu MHz` to be around `1200` for all the cores. Shouldn't it be 2100? Well, this question isn't about RyzenAdj.</sub>

Sometimes I want to keep my system as cool as possible. To achieve that I tried to set the maximum SoC Clock Frequency to 1200MHz or to 2100MHz.

```
$ sudo ryzenadj --max-socclk-frequency=1200
Sucessfully set max_socclk_freq to 4b0
```
```
$ sudo ryzenadj --max-socclk-frequency=2100
Sucessfully set max_socclk_freq to 834
```

In both cases the command is successful, but it doesn't seem to have any effect: `/proc/cpuinfo` shows that the `cpu MHz` can rise up to 3700MHz when the system is under load.

I haven't tried any of the other options of RyzenAdj because I wouldn't know what values to pass them. I don't know whether they would work or not.

`sudo ryzenadj -i` doesn't show any information at all. Is this normal?

Am I doing anything wrong?
What else can I try in order to force my system to stay as cool as possible?

FlyGoat, 2019-11-15:
SoC clock frequency doesn't mean CPU frequency. It's probably means internal bus frequency.

For now we can only say current/temperature related options are valid on Picasso based processors. You can try to limit stapm and ppt power to archive power saving.

furkanmustafa, 2020-03-23:
`/proc/cpuinfo` shows the actual current frequency.

you can check the following for "minimum" frequency.

```
cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_min_freq
```

when you want to keep it to minimum as much as possible, you can use `powersave` or `conservative` scaling governors.

```
echo conservative | tee /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor
```

you don't need RyzenAdj for any of these.

closed 2020-05-18 by FlyGoat

talalmash, 2020-05-24:
I can't figure out any way to raise the minimum frequency on 3500u on Windows. Playing with SoC frequency actually lowered idle frequency, donno how to raise it.
Thanks

dracula99gr, 2025-09-15:
Hello! Ryzen 3 3200u user here! 

I managed to successfully do that using cpupower-gui via commands (i had to manually compile it and install few extra packages)

sudo cpupower-gui freq --min 2600 --max 2600 0-3
&
sudo cpupower-gui -p
for performance governor

But i got another issue... The max value I can set it limited to the bax base clock. Is it possible to use max boost frequency with any tool?

I already tried changing tdp to the max of my cpu via
sudo /usr/local/bin/ryzenadj --stapm-limit=25000 --fast-limit=25000 --slow-limit=25000 --tctl-temp=90 --max-performance

But sadly it does not magically allow setting more than base...
I am trying to find more info online to no avail....

Also I suffer from no info issue

udo /usr/local/bin/ryzenadj -idetected compatible ryzen_smu kernel module
CPU Family: Picasso
SMU BIOS Interface Version: 12
Version: v0.18.0 
refresh_table failed
Unable to init power metric table: -5, this does not affect adjustments because it is only needed for monitoring.