FlyGoat/RyzenAdj #39: Build return an error

RemyL, 2019-11-15:
I followed instructions to build Ryzenadj but it doesn't work. I may have missed something or i need more dependencies to build on fedora.

First, i'm using Fedora 31 stable. i installed make, get and libpciaccess-devel-0.15-2.fc31.x86_64 (non-devel are also installed).

I get this error during building:
fatal error: pci/pci.h : No such file or directory
   71 | #include <pci/pci.

make[2]: *** [CMakeFiles/ryzenadj.dir/build.make:63: CMakeFiles/ryzenadj.dir/lib/libpci.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/ryzenadj.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

I guess it's an error linked to libpci but i don't understand why because dependencies are installed

Thanks

FlyGoat, 2019-11-15:
libpci Probably named pciutils-devel on Fedora?

RemyL, 2019-11-15:
Thanks, it needed pciutils-devel. Build worked perfectly

closed 2019-11-15 by RemyL