FlyGoat/RyzenAdj #42: pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config

MasterCATZ, 2019-12-16:
Ubuntu 19.10 
Kernel 5.4.3-050403-generic
libpci-dev is already the newest version (1:3.6.2-2)
Ryzen 3900x 

as root 
ryzenadj --max-socclk-frequency=2000 --tctl-temp=75 --fast-limit=50000 --slow-limit=30000

pcilib: Cannot open /sys/bus/pci/devices/0000:00:00.0/config

FlyGoat, 2019-12-16:
Sorry but I have nothing to do with this issue.
I had received some report about this issue on Ubuntu, but it didn't reproduced on other distros.
My personal opinion is that Ubuntu's kernel have some extra hardened protection mechanism preventing applications access PCI Config Space but can't find neither what is it nor how to disable it.

MasterCATZ, 2019-12-16:
Ok I might try uninstalling app-armor or something someone said you can enable more pci options in the kernel so i might need to compile my own 

this is a copy of the config file 
[config.zip](https://github.com/FlyGoat/RyzenAdj/files/3967807/config.zip)

so if I solve this issue would i be able to set the PPT of my 3900x on the fly ?

Yrahate7, 2020-03-02:
@MasterCATZ, disable secure boot from bios n then try Again

mirh, 2020-12-01:
Aside of secure boot, could it be this?
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1818816

commit 9a5d40d1 (2021-02-24): add documentation for all power, time and temperature options based on Renoir
add FAQ
resolves #25
resolves #41
resolves #42
resolves #48
resolves #49
resolves #50
resolves #53
resolves #58
resolves #72
closed 2021-02-24 by FlyGoat