FlyGoat/RyzenAdj #44: Threadripper 3 support

justanerd, 2020-01-26:
I tried running ryzenadj on my Threadripper system:
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xffffffff, arg0: 0xffffffff, arg1:0xffffffff, arg2:0xffffffff, arg3:0xffffffff, arg4: 0xffffffff, arg5: 0xffffffff
Faild to get SMU: (null), test message REP: ffffffff
Unable to init ryzenadj, check permission
Any way to get this supported? 
I would help with testing etc.

FlyGoat, 2020-01-26:
That means SMU have totally no response. Probably TR3 have different SMU mailbox address/calling convection?

FlyGoat, 2020-01-26:
You can try this tool and see if you can get anything.

https://github.com/irusanov/SMUDebugTool

Probably @irusanov can give you some assistant. I didn't follow the newer SMU firmware but he did some further research.

irusanov, 2020-01-26:
ZenStates works with Threadripper 3 series on TRX40 motherboard, so the addresses and commands are the same as on Matisse. However, if you need a linux tool, you're out of luck at the moment.

Also, some things have changed along the way and I haven't updated the app yet. 

EDC = TDC = PPT = 0 was working before (as ignored), while you can now actually set 0 for all of them, which limits the CPU to the lowest possible frequency. So if using the Auto OC feature, you need to set those values to something that makes sense. The default 0 in the app won't work. 

The thing is Auto OC is not working at all, because whatever parameter you change, the CPU just ignores it and continues to do its thing. P-States don't work - you can read/write them without any effect. Maximum boost frequencies have no effect. The only thing you can do is manual overclock - it still works.

justanerd, 2020-01-26:
That doesn't sound good.
ZenStates seems to work at least somewhat, a lower frequency is applied but I can't verify the vcore really.
zenstates.py  -p2 --enable -f 40 -dF -v 99
After that the frequency goes down to 800mhz

Setting the PPT would be really nice so I can ran a high default and a lower one if I don't need It.
Sadly AMD isn't really supporting linux especially overclocking etc.
Restarting the PC everytime I wan't to change my PPT is a little to much hassle.
My board is a TRX40 AORUS MASTER don't know if that makes a difference.

irusanov, 2020-01-26:
That zenstates (for linux) is a different tool and only supports changing P-States (basically rewriting the MSR responsible for that) but they don't work (as expected) on newer gen.
I was using that myself with ryzen 1800X, but it is almost useless for 3rd gen.
As I said, you're out of luck for the moment. Some of these tools need to be updated, but that would only work for manual overclock. It is possible to set PPT and other limits with smu commands, but no linux tool supports it atm.

justanerd, 2020-01-26:
Any way to get some progress? Maybe crowdfunding or something? I would certainly help a bit. But I guess the count of people using Linux and Threadripper is rather low. I will post on the level1forum maybe this will help.

FlyGoat, 2020-01-26:
> Any way to get some progress? Maybe crowdfunding or something? I would certainly help a bit. But I guess the count of people using Linux and Threadripper is rather low. I will post on the level1forum maybe this will help.

The best help is give AMD guys some pressure and call them to opensource AGESA again. No more reverse engineering!

irusanov, 2020-01-26:
I will try to add some commands to that linux version, setpci should work ok. Possible things to add is "enable/disable" oc mode, set edc, ppt, tdc, all core frequency in oc mode, oone core frequency in oc mode, overclock vid

justanerd, 2020-02-15:
> I will try to add some commands to that linux version, setpci should work ok. Possible things to add is "enable/disable" oc mode, set edc, ppt, tdc, all core frequency in oc mode, one core frequency in oc mode, overclock vid
> 
> PS: Ok, good news. It was quite easy to add functions to send commands to the SMU and I'm now able to use all known commands. Will add some useful ones and will commit to my fork of the ZenStates-Linux.

Works for me .
thank you very much

PR #98 (2021-03-03): new error msg: Only Ryzen Mobile Series are supported
resolves #97
resolves #44

avoid questions about Desktop support
commit 1f70813a (2021-03-03): new error msg: Only Ryzen Mobile Series are supported
resolves #97
resolves #44
closed 2021-03-03 by FlyGoat