FlyGoat/RyzenAdj #45: HP envy X360 Ryzen5 2500u doesn't work

SlintK, 2020-02-12:
HP-ENVY-x360-Convertible-15-bq1xx:~/RyzenAdj/build$ sudo ./ryzenadj -i
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
pcilib: sysfs_write: write failed: Operation not permitted
Not a Ryzen NB SMU, BIOS Interface Ver: 0x1Unable to init ryzenadj, check permission

FlyGoat, 2020-02-17:
Try turn of secure boot.

SlintK, 2020-02-17:
Hello, 

Thanks for help! It worked!

closed 2020-02-17 by SlintK