FlyGoat/RyzenAdj #46: Renoir support. Offering help

koriwi, 2020-04-22:
Haven't read anything about Renoir here.
My laptop should arrive this or next week. If I can help (mainly using Linux) let me know.

FlyGoat, 2020-04-23:
Sorry but I'm not going to reverse engineering Renior's SMU stuff because lack of time and information.

I'm happy to add Renior support if somebody can give me a list of Renior's SMU call.

koriwi, 2020-04-23:
No problem. 
I don't know if you know [this](https://github.com/irusanov/ZenStates) or how much it helps (because i don't know how much desktop will differ from mobile)

koriwi, 2020-04-27:
OK. I found this Video  where he talks about lenovos vantage software which allows to enable the performance mode. Which allows for 42 watts peaks and 37watts sustained tdp.
Timestamps 2:10 and 3:38
 https://youtu.be/3jf7itO2YKg

Is there a way to "easily" catch the smu calls?

koriwi, 2020-05-02:
For everyone finding this. Reverse engineering seems to be somewhat complete: https://github.com/FlyGoat/ryzen_nb_smu/issues/3

FlyGoat, 2020-05-03:
I'm going to do sth with the effort when I have time:-)

irusanov, 2020-05-03:
It seems to share the same SMU addresses as other APUs. Available commands are rather limited though. It would be interesting if the trick with P-States still works.

koriwi, 2020-06-04:
Ok, after waiting almost 7 weeks, my laptop arrived. Is there something i should test?

FlyGoat, 2020-08-20:
> 
> 
> Ok, after waiting almost 7 weeks, my laptop arrived. Is there something i should test?

Now Renoir is supported. I'm goting to close this issue.

closed 2020-08-20 by FlyGoat