FlyGoat/RyzenAdj #47: 3700u Dell settings don't last

Tritonn204, 2020-04-25:
I was wondering if anyone else is using this tool on a Dell Inspiron with a Picasso processor.
uProf on my system is showing that whenever I apply my parameters, they don't even last for a second. I have no clue what keeps resetting the settings.

If anyone has tips or ideas please do share them.

ThinkAndFly, 2020-05-02:
Flygoatf https://www.reddit.com/r/Amd/comments/anbl95/ryzen_master_for_mobile_ryzen_modify/

> 
> ABOUT RYZENADJ BUG
> 
> Some users reported that the settings of RyzenAdj aren't constant. The power limits will be pulled back to 15W or some other values after some time.
> 
> Well, that's not a problem with RyzenAdj itself but probably your ACPI table or EC or AMD SMU itself. Since the RyzenAdj is doing basically identical with what DSDT injection did. There must be something that is fighting against the RyzenAdj.
> 
> Since I just figured out how to determine current settings just as what uProf did, I'll try to make RyzenAdj as a daemon running background and check the value frequently to see if it is changed by others, if so, then set it back to target value.
> 
> Also, I think it may possible to solve the issue from the root. Just like edit your ACPI tables to prevent EC Query a method that can adjust thermal settings.
> 
> If you encounter that problem frequently, please let me know and give me your original DSDT, I'm still trying to locate the root of the the issue.

FlyGoat, 2020-05-18:
Thanks for replying. I'm going to close this issue.

closed 2020-08-20 by FlyGoat

mirh, 2020-12-02:
https://www.techpowerup.com/forums/threads/hp-envy-x360-2500u-power-throttling-question.253853/page-2#post-4195256
According to this guy and my actual experience, HP Command Center seems somehow able to avoid this from happening.