FlyGoat/RyzenAdj #48: WinRing0 Err: 0x2Unable to get PCI Obj

tarragon777, 2020-04-30:
WinRing0 Err: 0x2Unable to get PCI Obj
Unable to init ryzenadj, check permission
Help pls.
What can it be? Same error on a all version in gui
:(

FlyGoat, 2020-05-03:
Which OS?

tarragon777, 2020-05-03:
> Which OS?

Windows 10

FlyGoat, 2020-05-04:
You'd better try to disable security features of Windows10.

tarragon777, 2020-05-04:
> You'd better try to disable security features of Windows10.

i don`t know what happened but after booting OS today Gui version ryzenAdj 1.14.1 works fine now. Same issue stays on newer versions 2.00, 2.1.0, 2.2.0. Tried  to give rules for all users and run as admin, switched off windows defender etc. All these things really changed nothing. Will be using older version of GUI

crimsoncandy, 2020-05-11:
Check if you have enabled Windows Defender> Device Security > Core Isolation > Memory Integrity. ryzenadj doesn't work with this enabled with same issue

commit 9a5d40d1 (2021-02-24): add documentation for all power, time and temperature options based on Renoir
add FAQ
resolves #25
resolves #41
resolves #42
resolves #48
resolves #49
resolves #50
resolves #53
resolves #58
resolves #72
closed 2021-02-24 by FlyGoat