FlyGoat/RyzenAdj #50: How to restore the original value after modification

zhangyiming748, 2020-05-08:
I ran the demo on Windows, I just wanted to try it out, how to restore it to the original value now? I feel the computer's performance is significantly reduced

FlyGoat, 2020-05-08:
Performance a reboot would reset all settings.

zhangyiming748, 2020-05-08:
Do you mean that restarting the computer will restore all settings? Sorry my english is not very good, thanks for reply

FlyGoat, 2020-05-08:
> Do you mean that restarting the computer will restore all settings? Sorry my english is not very good, thanks for reply

Exactly. All settings are volatile.

stopyan, 2020-05-09:
是的，相关操作在电脑重启后初始化为主板bios默认值。

mirh, 2020-12-02:
Couldn't there be a `--reset` parameter?
Ryzencontroller has a button for that for example.

Falcosc, 2021-02-23:
It's the same problem like: https://github.com/FlyGoat/RyzenAdj/issues/57

Not possible to read out default values, so this would only possible if you store things which was done before.

This feature does only make sense for UI Integrations like Ryzen Controller or https://github.com/sbski/Renoir-Mobile-Tuning

commit 9a5d40d1 (2021-02-24): add documentation for all power, time and temperature options based on Renoir
add FAQ
resolves #25
resolves #41
resolves #42
resolves #48
resolves #49
resolves #50
resolves #53
resolves #58
resolves #72
closed 2021-02-24 by FlyGoat