FlyGoat/RyzenAdj #52: Ryzen 4000 series compatibility

Freekers, 2020-05-12:
Is RyzenAdj already compatible with the new Ryzen 4000 series?

Thank you!

FlyGoat, 2020-05-18:
It is known that TDP function works with Renoir.

adam900710, 2020-07-18:
Doesn't look like the case, at least for Ryzen Pro 4750U

I set the TDP to 25W/35W, but during kernel building (make -j20), no difference in CPU temp (~60C), nor the completion time
(<6min for the localmodconfig), nor the CPU freq (around 2.1~2.2Ghz for all cores).

While for another laptop, using 4800U, its TDP is much higher (25W I guess?), it finishes kernel compile faster and all cores runs around 3Ghz for all core loads.

Did I miss something or the default setup of Ryzen Pro 4750U is already 25W/35W, not the 15W claimed?

adam900710, 2020-07-18:
Another try, now 30/45W.

No difference in CPU freq, temp, nor kernel compile time.

Does this mean Ryzen Pro lineup has a different config? Or the upper limit is harded coded in certain lineup?

What info can I provide to debug?

FlyGoat, 2020-07-18:
@adam900710 I just got a 4800U laptop (Yoga 14s) and will do some test in next few days.

Thanks.

adam900710, 2020-07-18:
That's great.

It would be even better if the current and power monitoring can be integrated into k10temp module (just like the desktop ryzen 3000).

bezirg, 2020-07-18:
> That's great.
> 
> It would be even better if the current and power monitoring can be integrated into k10temp module (just like the desktop ryzen 3000).

I think k10temp renoir support is supposed to come with kernel 5.8. not 100% sure though

adam900710, 2020-07-19:
> > That's great.
> > It would be even better if the current and power monitoring can be integrated into k10temp module (just like the desktop ryzen 3000).
> 
> I think k10temp renoir support is supposed to come with kernel 5.8. not 100% sure though

It's just temp monitoring (Tctl and Tdie) in 5.8. In comparison, the desktop ryzen 3000 has Vcore, Vsoc, Tctl, Tdie,Tccd1, Tccd2 (for two ccd chips), Icore and Isoc (which can be used to calculate the Pcore and Psoc)

FlyGoat, 2020-07-19:
> It's just temp monitoring (Tctl and Tdie) in 5.8. In comparison, the desktop ryzen 3000 has Vcore, Vsoc, Tctl, Tdie,Tccd1, Tccd2 (for two ccd chips), Icore and Isoc (which can be used to calculate [...]

That will need extra effort on reverse engineering SMN sensor registers and currently out of my scope.

moson-mo, 2020-08-12:
> Ryzen Pro 4750U

Hmm. Working fine here on a T14s with that CPU (at least on AC).

Which NB do you have? Maybe there are some limits applied by the EC as well. (I had that issue with my previous T480s where I had to trick the EC too; with an Intel CPU though)

siu00adg, 2020-08-12:
Same, T14s 4750U working great for me. Weirdly Geekbench only scores well (74xx multi) on AC and on battery it falls to 5xxx. When I run my own code (e.g. a numpy neural network) it runs like a beast on battery and AC.

adam900710, 2020-08-12:
It's now working after commit d5e03022844edfb44dc70ad493313b5b5d1f39fc.

closed 2020-12-20 by FlyGoat