FlyGoat/RyzenAdj #53: Temperature Limit just does not work

vvgtomasovic, 2020-05-15:
Hi,

Issue explanation
Temperature Limit just does not work, whenever temperature hits 75 Celsius CPU will downclock to 2.00 to get temps down, and as soon as its lower than 75 it will work on boost clock again.
Any way to fix this? Everything else works fine.

System information
Acer Nitro 5 AN515-42, BIOS v1.16
AMD Ryzen 5 2500u
RX 560X 4GB
12GB RAM
Windows 10 Pro with latest update

[+] I already checked for other issues

[+] Ryzen Controller version: 2.2.0

I'm using

[+] Windows (.exe package)

FlyGoat, 2020-05-18:
You should blame laptop vendor/AMD for this.

vvgtomasovic, 2020-05-18:
> You should blame laptop vendor/AMD for this.
> ACPI/EC is puling the limit.

Damn I kinda knew this is a hw issue but I was hoping that it could be solved..
I guess BIOS update by Acer is only sollution?

FlyGoat, 2020-05-18:
> > You should blame laptop vendor/AMD for this.
> > ACPI/EC is puling the limit.
> 
> Damn I kinda knew this is a HW issue but I was hoping that it could be solved..
> 
> I guess BIOS update by Acer is only solution?

Probably ACPI edit or other hacks can disable this limit..

vvgtomasovic, 2020-05-18:
> > > You should blame laptop vendor/AMD for this.
> > > ACPI/EC is puling the limit.
> > 
> > 
> > Damn I kinda knew this is a HW issue but I was hoping that it could be solved..
> > I guess BIOS update by Acer is only solution?
> 
> Probably ACPI edit or other hacks can disable this limit..

Do you recommend any Tool or a video/forum tutorial on how to do that?

vvgtomasovic, 2020-05-18:
> > > You should blame laptop vendor/AMD for this.
> > > ACPI/EC is puling the limit.
> > 
> > 
> > Damn I kinda knew this is a HW issue but I was hoping that it could be solved..
> > I guess BIOS update by Acer is only solution?
> 
> Probably ACPI edit or other hacks can disable this limit..

Tried the ACPI edit, even found a way that Works for some people. But no luck on my Laptop. 
Temperature limit cannot be removed from ACPI

commit 9a5d40d1 (2021-02-24): add documentation for all power, time and temperature options based on Renoir
add FAQ
resolves #25
resolves #41
resolves #42
resolves #48
resolves #49
resolves #50
resolves #53
resolves #58
resolves #72
closed 2021-02-24 by FlyGoat