FlyGoat/RyzenAdj #54: Renoir Support

FlyGoat, 2020-05-18:
There are multiple issues (#46 #52) asking about Renoir support status.

This issue will be the meta issue used to track Renoir support.

Btw:
Personally I'm not interested in bringing RyzenAdj to desktop systems but any contribution is welcomed.

adam900710, 2020-07-19:
Any recommendation for someone who wants to investigate into the whole SMU situation?

siu00adg, 2020-07-30:
Has anyone tested the pull request from @sbski ?

FlyGoat, 2020-07-30:
> Has anyone tested the pull request from @sbski ?

Can confirm it works, I've merged it but need some trick to fix the CI and publish a release.

sbski, 2020-08-05:
> Any recommendation for someone who wants to investigate into the whole SMU situation?

Hit me up at keaton.blomquist@gmail.com! I am investigating quite a lot about the SMU and there are some really cool applications that people are creating with what we know about the SMU so far.

siu00adg, 2020-08-05:
Thanks for your work @sbski - I cranked my ThinkPad T14s (Ryzen 4750U Pro) to 45W and it crushed geekbench: https://browser.geekbench.com/v5/cpu/3172140

Only seems to work when plugged in though, performance is much worse while on battery, I'm thinking it might be something hard-coded in the bios.

sbski, 2020-08-05:
> Thanks for your work @sbski - I cranked my ThinkPad T14s (Ryzen 4750U Pro) to 45W and it crushed geekbench: https://browser.geekbench.com/v5/cpu/3172140
> 
> Only seems to work when plugged in though, performance is much worse while on battery, I'm thinking it might be something hard-coded in the bios.

There's a lot that can effect the power limits and its hard to nail them all down. My G14 doesn't really complain about it on battery weirdly enough. I can really shove the power draw down with even lower power limits.

Falcosc, 2021-02-23:
@adam900710 I would like to know if you can confirm my findings: https://github.com/FlyGoat/RyzenAdj/wiki/Renoir-Tuning-Guide

bezirg, 2021-02-23:
@Falcosc @adam900710 Have you experimented also with the max/min: vcn, fclk, gfxclk adjustable settings? Do they make any difference or are they silently ignored for renoir as is for stapm?

Falcosc, 2021-02-23:
We don't have message Ids for them. The old message Ids don't fit and the known message Id offset compared to Picasso didn't work either. 
Next release will tell you that these are unsupported.

I did try most of the message Ids with the known good values for frequency to down clock to a known value during load. Nothing did happen, but during this painful process I found 2 undocumented messages for Reinor 🥳
The new messages will be included in the next release as well.

PR #113 (2021-03-10): Document which Processors are supported and which commands do work
New Support Matrix in https://github.com/FlyGoat/RyzenAdj/wiki/Supported-Models

Resolves #28
Resolves #30
Resolves #33
Resolves #54
Resolves #80
commit 04aa9560 (2021-03-10): Document which Processors are supported
Resolves #28
Resolves #30
Resolves #33
Resolves #54
Resolves #80
closed 2021-03-10 by FlyGoat