FlyGoat/RyzenAdj #57: Ryzenadj --info

dcarusil, 2020-06-25:
Hello,

I was only wondering about the -i argument. I see no output and was curious if it is implemented? My understanding is it will gather current settings from the device, is it possible to gather OEM default settings? Or is it impossible to read the registers/memory?

Steps;
Restart pc
Open admin cmd
ryzenadj --info

Thank you.

dcarusil, 2020-06-26:
I realize amd uprof can capture default stapm and such. I am just curious :)

Falcosc, 2021-02-23:
You can not read the default values. You can only read active values. Tools like uprof does store the active value as default value during initialization.

Since RyzenAdj does not keep running, we would lose the last extracted active value after each call of RyzenAdj.

commit 59ae793a (2021-03-24): Add Ptable Support
show power monitoring values after adjust with --info
outputs the whole table before and after adjust with --dump-table for debugging
resolves #57
commit 62c6fd49 (2021-03-24): Add Ptable Support
show power monitoring values after adjust with --info
outputs the whole table before and after adjust with --dump-table for debugging
resolves #57
commit a8a8de13 (2021-03-25): Add Ptable Support
show power monitoring values after adjust with --info
outputs the whole table before and after adjust with --dump-table for debugging
resolves #57
commit 6daa8f4b (2021-04-01): Add PM Table Support
show power metric values after adjust with --info
outputs the whole table before and after adjust with --dump-table for debugging
resolves #57
closed 2021-04-01 by FlyGoat