FlyGoat/RyzenAdj #58: Ryzen 3550H Manjaro

sticsk, 2020-07-07:
Just want to let people know that on Ryzen 3550H on Manjaro with TDP 35W works flawlessly using these settings

`./ryzenadj --stapm-limit=35000 --fast-limit=35000 --slow-limit=35000 --tctl-temp=90` tested it by watching the core speeds and stress testing with `s-tui` and also playing Battlefield 1, and both seem to use the CPU to its full speed(~3.7GHz) and not to what used to be maximum (2.1GHz)

mthw0, 2020-12-20:
@sticsk Since you mentioned you have a 3550H and I do too, does your laptop allow you to control fan speeds? Or do you know a tool that can do it? I have been looking for something for a while now and haven't found anything that works.

sticsk, 2020-12-20:
Yeah I use faustus.

sticsk#0990 on discord if u want help or something

closed 2021-02-24 by FlyGoat

Corrupter-rot, 2021-08-23:
@sticsk Can you provide your Correct Discord ID as there are some question i need to ask you??

sticsk, 2021-08-23:
> @sticsk Can you provide your Correct Discord ID as there are some question i need to ask you??
> 
> 

Discord : sticsk#0990
Telegram : @The_Cryptarch

I'd rather you to message me on Telegram since I barely check Discord, but if you contact me in upcoming days I promise i'll check it 🤣

Corrupter-rot, 2021-09-12:
@sticsk Search for your Telegram username can't found one 

Telegram: [K4R7IK_Ku5hw4h4](https://t.me/K4R7IK_Ku5hw4h4)
Discord: Corrupter#8885

Here is my contact info 
Add me

commit 9a5d40d1 (2021-02-24): add documentation for all power, time and temperature options based on Renoir
add FAQ
resolves #25
resolves #41
resolves #42
resolves #48
resolves #49
resolves #50
resolves #53
resolves #58
resolves #72