FlyGoat/RyzenAdj #61: CPU/GPU locking to minimum on 2500u

crimsoncandy, 2020-08-07:
with new ryzenadj v0.6.2, after applying any settings
CPU locks to 400MHz (Task Manager)
GPU locks to 200MHz (AMD OSD)
Memory locks to 400MHz (AMD OSD)
Basically everything goes to their minimum possible value, reducing performance, it doesn't matter what settings I apply.
There are no errors shown.

OS: Windows 10 pro 1909
CPU: Ryzen 5 2500u

FlyGoat, 2020-08-08:
Is it a regression of v0.6.2, or did it happen on previous versions?

sbski, 2020-08-08:
No that was not intentional

FlyGoat, 2020-08-08:
> No that was not intentional

Oops, to confirm, serivce id on Raven is `0x1f` and on Picasso is `0x1a`?

I'll release a emergency hotfix for this.

sbski, 2020-08-08:
I got it fixed! No, Raven and Pocasso should both be 0x1F. I think I just made a typo when I was adding support for Renoir. I got a new build and updated the code already. @crimsoncandy 0.6.3 has the fix for it. Let me know if it works

crimsoncandy, 2020-08-08:
@sbski I tested version 0.6.3, it is working now.

closed 2020-08-08 by crimsoncandy

sbski, 2020-08-08:
> @sbski I tested version 0.6.3, it is working now.
Great!  I am looking into adding in some more features here sometime.