FlyGoat/RyzenAdj #68: Values always successfully written but not applied sometimes

kerneltoast, 2020-09-24:
I noticed that when I try to run RyzenAdj on boot to apply my desired settings, my settings are successfully written but not applied. However, when I run RyzenAdj myself in a terminal, my settings do get applied, but not always; sometimes just running RyzenAdj in the terminal does nothing. Also, applying RyzenAdj doesn't do anything while on battery (the values are written but do nothing; they are still not applied after plugging the charger in).

This is on a Tongfang laptop with Ryzen 7 4800H. Any suggestions on how to at least make RyzenAdj apply successfully on boot?

lazerl0rd, 2020-10-26:
I'm experiencing a similar issue on my Zephyrus G14 on Windows 10. On boot, I get values such as 45W and 55W being set - but when running the script manually I successfuly have 65W being set.

Benman2785, 2021-01-20:
1. scheudule a task on every login
2. write a batch that sets your wished settings
3. in that batch "wait" 15s before you apply the commands of RyzenAdj

boom - its now working fine ;)

Falcosc, 2021-02-23:
RyzenAdj is not the only piece of code which can send commands.

Anything with enough permissions can change your limits.

For example the Lenovo firmware does set limits during Power source switch and from time to time based on an unknown reason.

You need to reapply the limits every time after something else did change it. You can not prevent your device firmware from overwriting your limits. You should be grateful that your firmware does allow you do change these limits at all :)

I use HWiNFO alerts to execute RyzenAdj from time to time because HWiNFO is already running for monitoring anyway.
If you don't already monitor your device I would recommend using https://github.com/sbski/Renoir-Mobile-Tuning instead of HWiNFO

lazerl0rd, 2021-02-24:
> RyzenAdj is not the only piece of code which can send commands.
> 
> Anything with enough permissions can change your limits.
> 
> For example the Lenovo firmware does set limits during Power source switch and from time to time even without power source switch based on an unknown reason.
> 
> You need to reapply the limits every time after something else did change it. You can not prevent your device firmware from overwriting your limits. You should be grateful that your firmware does allow you do change these limits at all :)
> 
> I use HWiNFO alerts to execute RyzenAdj from time to time because HWiNFO is already running for monitoring anyway.
> If you don't already monitor your device I would recommend using https://github.com/sbski/Renoir-Mobile-Tuning instead of HWiNFO

Both Sultan and I had issues with RyzenAdj on Linux, not Windows, as suggested by https://github.com/FlyGoat/RyzenAdj/issues/90.

Falcosc, 2021-02-24:
Only crone jobs or PTable monitoring will help in Linux.

I can't help with automation in Linux, because I don't know enough to tell how to handle power source change events or which tools can be used to monitor PTable.

Event if we want to develop a monitoring tool it would be not included into RyzenAdj instead it would be a new tool which does call RyzenAdj.

But event monitoring does call your SMU every second, so monitoring is as good as creating a cronejob which does apply the limits every minute. So why do you dislike Benman2785 suggestion if monitoring tools do basically the same thing?

lazerl0rd, 2021-02-24:
> Only crone jobs or PTable monitoring will help in Linux.
>
> I can't help with automation in Linux, because I don't know enough to tell how to handle power source change events or which tools can be [...]

Unlike your insightful comment, Benman2785's suggestion didn't really help with anything. The problem that Sultan and I experienced is two-fold:

1) Sometimes RyzenAdj "does nothing", while in most cases it just works. The issue was mainly opened for this reason.
2) For some laptop models, it seems that RyzenAdj isn't able to set anything on battery (or maybe it's being overridden as you mentioned).

Benman2785's isn't helpful because as long as 1) occurs, there's no assurance that the script will ever succeed. For 2), he didn't even touch on the issue. This issue is also persistent across Windows (tested by me) and Linux (tested by Sultan and I), and while I could just write a Shell script for Linux, that isn't going to help much (as well as it being common sense to try). It's a second try at rolling the dice if RyzenAdj will actually do anything or not.

Falcosc, 2021-02-24:
There are 4 usecases which do look like RyzenAdj do nothing:
- RyzenAdj response ok, change in PTable visible and have effect on your device
- RyzenAdj response ok, change in PTable visible and have no on your device
- RyzenAdj response ok, change in PTable visible and have effect on your device but get overwritten later
- RyzenAdj response not ok

Which of them is true for your issue?

If we implement PTable we could only check if change is visible PTable, but we don't no if it does work, or if it does get overwritten later or if another limit is holding you back

lazerl0rd, 2021-02-24:
> There are 5 use-cases which do look like RyzenAdj does nothing:
> 
> * RyzenAdj response ok, change in PTable visible and have no effekt your device
> * RyzenAdj response ok, change in PTable visible and have effect on your device, but get overwritten later
> * RyzenAdj response ok, change in PTable visible and have effect on your device, but another limit is holding you back
> * RyzenAdj response ok, change in PTable not visible
> * RyzenAdj response not ok
> 
> Which of them is true for your issue?
> 
> If we implement PTable we could only check if change is visible in PTable, but we don't know if it does work, or if it does get overwritten later or if another limit is holding you back.
> 
> At which part of the use-case description do you have trouble finding out which case is true for your device?

I no longer have access to my Ryzen laptop, as I ended up returning it, but from what I remember the problem seemed to be either "RyzenAdj response ok, change in PTable not visible" or "RyzenAdj response ok, change in PTable visible and have no effekt your device" and most likely the former.

Falcosc, 2021-02-24:
I did try to describe the solution for anybody in a more generic way.

https://github.com/FlyGoat/RyzenAdj/wiki/FAQ#temperature-or-power-limit-does-not-work

What would you like to add there?

PR #110 (2021-03-11): Add instructions for installation
Resolves #64
Resolves #68
Resolves #89
commit d7da2d6d (2021-03-11): Add instructions for installation
Resolves #64
Resolves #68
Resolves #89
closed 2021-03-11 by FlyGoat