FlyGoat/RyzenAdj #71: Error compiling RyzenAdj binary in Fedora 33

GrapplingHooks, 2020-11-05:
>$ make
Scanning dependencies of target libryzenadj
[  8%] Building C object CMakeFiles/libryzenadj.dir/lib/libpci.c.o
In file included from /home/deiram/RyzenAdj/lib/libpci.c:5:
/home/deiram/RyzenAdj/lib/nb_smu_ops.h:71:10: fatal error: pci/pci.h: No such file or directory
   71 | #include <pci/pci.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/libryzenadj.dir/build.make:82: CMakeFiles/libryzenadj.dir/lib/libpci.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:97: CMakeFiles/libryzenadj.dir/all] Error 2
make: ** [Makefile:103: all] Error 2

TomAmpiath, 2020-11-18:
Same here (fedora 33) with libpciaccess-devel installed.

`fatal error: pci/pci.h: No such file or directory`

TomAmpiath, 2020-11-22:
Install pciutils-devel on fedora:

`sudo dnf install pciutils-devel`

closed 2020-12-20 by FlyGoat