FlyGoat/RyzenAdj #72: Question about STAPM terminology

tobiasjakobi, 2020-11-05:
Hello,

first of all thanks to the author for this wonderful tool. I'm using it on a Dell D5 5505, operating system is Gentoo Linux.

My question is about the STAPM terminology used in ryzenadj. If you take a look at [Ryzen Controller](https://www.ryzencontroller.com/#config), they have a graph displaying how STAPM basically works. Three TDP values are used, the CPU TDP a, the long boost TDP b and the short boost TDP c. It's obvious that only a configuration satisfying b > c > a makes sense.

Now ryzenadj has different terminology, the sustained power limit, fast PPT power limit, and slow PPT power limit.

What I'd like to know is, how these terms map to each other. I figure that the CDU TDP ist the sustained power limit. But what is the fast PPT power limit? Is it the value for the long boost, or the short one?

And when I'm already at it, what is the reason behind the lack of reading the configuration back from the SMU? I figure it has something to do with TransferTableSmu2Dram not working as expect?

Thanks in advance!

With best wishes,
Tobias

FlyGoat, 2020-12-20:
> Hello,
>
> first of all thanks to the author for this wonderful tool. I'm using it on a Dell D5 5505, operating system is Gentoo Linux.
>
> My question is about the STAPM terminology used in ryzenadj. If [...]

Well, we don't know either. Presumably fast PPT is short boost, slow PPT is long boost, stapm is sustained power.

But without official explanation we can't do much thing to explain it further.

> 
> And when I'm already at it, what is the reason behind the lack of reading the configuration back from the SMU? I figure it has something to do with TransferTableSmu2Dram not working as expect?

Yes, it fails constantly.

> 
> Thanks in advance!
> 
> With best wishes,
> Tobias

tobiasjakobi, 2020-12-20:
Thanks, that already helps a lot. E.g. I currently use these values:  
`--stapm-limit=45000 --fast-limit=50000 --slow-limit=47000`

Hence we have a = 45W, b = 47W, c = 50W, which would not satify b > c > a. So I should probably swap b with c, i.e.  
`--stapm-limit=45000 --fast-limit=47000 --slow-limit=50000`

-- Tobias

mirh, 2020-12-20:
https://forums.anandtech.com/threads/how-do-you-think-bristol-ridge-will-do-as-a-mobile-processor-compared-to-carrizo.2459807/page-3#post-37953816
https://www.amd.com/system/files/TechDocs/44065_Arch2008.pdf#page=271

Ehrm.. are you sure fast and slow, long and short, are actually even real terms?

tobiasjakobi, 2020-12-20:
@mirh: Are you asking me? I'm not sure what you mean by "real terms".

leoncryp, 2021-01-19:
Its been a while, but back few years ago when I was doing some test on my normie Acer laptop, if i remember right
I used AMD uProf to see detail stats, this is one of my configs:
--fast-limit=30000 --slow-limit=25000 --stapm-limit=23000 
you only need to worry about stapm-limit, that is your sustain speed,
fast-limit is something like short burst, it will keep it at that wattage depending on the temps.
The max i can push my laptop depends on its cooling capabilities, normal was 15 watt, but after i change it to liquid metal i was able to run 23watt consistent.
the slowest i can get it was 6 watt, cpu ran at 0.4 ghz. all cores when stressed.

Falcosc, 2021-02-23:
@leoncryp Be careful with this advice, because it isn't true for Renoir. I did create a Guide to tell about results of black box test approach: https://github.com/FlyGoat/RyzenAdj/wiki/Renoir-Tuning-Guide

tobiasjakobi, 2021-02-23:
@Falcosc Now that's some conclusive information. Thanks a lot for sharing!

commit 9a5d40d1 (2021-02-24): add documentation for all power, time and temperature options based on Renoir
add FAQ
resolves #25
resolves #41
resolves #42
resolves #48
resolves #49
resolves #50
resolves #53
resolves #58
resolves #72
closed 2021-02-24 by FlyGoat