FlyGoat/RyzenAdj #73: Unable to init ryzenadj even have admin permission

AlaricGilbert, 2020-11-12:
[image]
Cannot run example instruction even if I have admin permission
OS Windows10 2004
Core Isolation's Memory Integrity mentioned by other issues never opened.
[image]

AlaricGilbert, 2020-11-12:
build from latest commit also not working...
it generates a debug error: heap corruption
[image]

FlyGoat, 2020-11-13:
Maybe you have to disable secure boot and other security features.

The heap corruption is really wired. Actually I don't think RyzenADJ has anything to do with that. I'd recognise it as a WinRing0 issue. Any further debug information?

FlyGoat, 2020-12-20:
Closing, I'll reopen if there are further feedback.

closed 2020-12-20 by FlyGoat