FlyGoat/RyzenAdj #74: A litlle problem with my Hp

jpvmet0, 2020-11-27:
Hi, thanks a lot to build this, is an amazing tool.

I have an hp 15s-eq0262ng, Amd Ryzen 7, 3700u from Hp germany, it has from factory a block in the cpu to use only at 50 C, this cause that the cpu only give the 15% of the capacity. I tried to use ryzenAdj, but i cant. The bios advanced options are locked from HP

In windows, always said that the changes that i made with ryzen controller its acepted but i dont have any changes.
I tried in linux trought terminal and i recive this message.

Unable to init ryzenadj, check permission

can you have some fix?

mirh, 2020-12-01:
On linux you have to run the program with `sudo`

FlyGoat, 2020-12-20:
Closing as we can't do anything with it :-(

closed 2020-12-20 by FlyGoat