FlyGoat/RyzenAdj #76: Unable to apply ryzenadj

KonstantMan, 2020-12-15:
I installed [ryzen-controller-bin](https://aur.archlinux.org/packages/ryzen-controller-bin) from AUR , when i start the application i get an error  message saying  path to ryzenadj.exe is wrong 
so i change the path to /opt/Ryzen Controller/resources/app.asar.unpacked/build/bin/ryzenadj.exe or /usr/bin/ryzenadj
But when i change any settings it says Unable to apply ryzenadj

Thrashattack, 2020-12-16:
Same here: 

Operating System: Manjaro Linux
KDE Plasma Version: 5.20.4
KDE Frameworks Version: 5.76.0
Qt Version: 5.15.2
Kernel Version: 5.4.80-2-MANJARO
OS Type: 64-bit
Processors: 8 × AMD Ryzen 5 2500U with Radeon Vega Mobile Gfx
Memory: 6.8 GiB of RAM
Graphics Processor: AMD Radeon™ Vega 8 Graphics
With GTX 1050 and Ryzen 5 2500u 

Running Ryzen Controller from `/opt` dir as `sudo` isn't possible because of chrome's `--no-sandbox` flag. It's also not possible to pass the flag as an argument when calling Ryzen Controller. 

Console when apply: 
[image]

FlyGoat, 2020-12-20:
Please forward this to Ryzen Controller.

closed 2020-12-20 by FlyGoat