FlyGoat/RyzenAdj #77: [Question] Would it be possible to add fan control functionality?

mthw0, 2020-12-20:
I have a laptop with a 3550H CPU and there is no known way to adjust fan curves, or generally control fan speed manually. (One can only choose from existing profiles, but for me the quiet one is already too loud.) Would it be possible to gain access to fan speeds the same way you do with the other settings?

FlyGoat, 2020-12-20:
Well that's out of our scope. 
Different platforms have their own ways to control fans, we don't want to adopt thousands of platforms.
RyzenAdj intends to be utility to adjust CPU power management settings.

You may check nbfc and other tools.

Thanks.

closed 2020-12-20 by FlyGoat

BbIXOD, 2024-11-18:
I know you already answered this but looks like it is not so out of scope in my case. ryzen cpu literally not allows managing fan of it through ec registers/pwm and most likely it is managed via smu, so, i think, pretty much by same thing you are already working with (my model is ASUS Vivobook M1502IA).

(Doing this because looks like there is no way i will get my fans work on linux on par with cpu load myself in any way)