FlyGoat/RyzenAdj #78: Question: disable DRAM powersaving feature

Benman2785, 2020-12-30:
hi,

my AN515-42 (Acer Nitro 5) with Ryzen 2500U + RX560X got a "problem" with BIOS 1.16

they enabled dynamic RAM as powersaving feature - now my 16GB DDR4-2400CL16 clocks from 400MHz up to 1200MHz - and in some games it doesnt boost up to max speed (1200MHz CL16) (like in Cyberpunk2077 it is at 933MHz when playing - only when loading a savegame it reaches 1200MHz)

is there a way to force it to 2400MHz? or to disable that powersaving feature?

FlyGoat, 2020-12-30:
You can try to tweak FCLK related options.

Benman2785, 2020-12-30:
in what way?
make it dynamic aswell? like min 400MHz and max 1200MHz? does that work synced with ramspeed?

of force 1200 on min and max - will try that tomorrow ;)

Benman2785, 2021-01-07:
ok, can be closed

min/max at 1200MHz fixes RAM speed to 1200MHz - thank you ;)

Falcosc, 2021-02-27:
Looks like your platform support adjusting fabric clock. @Benman2785 please could you help me finding which metric is used to report the max speed, and which is used for the min speed?
Here is the description: https://github.com/FlyGoat/RyzenAdj/issues/94

Thank you

closed 2021-04-03 by Falcosc