FlyGoat/RyzenAdj #79: Question about vcn socclk and lclk

TechieAndroid, 2021-01-04:
What "values" do the vcn options take and what does it directly affect?
-s, --max-vcn=<u32>                   Maximum Video Core Next (VCE - Video Coding Engine) (Value)
-t, --min-vcn=<u32>                   Minimum Video Core Next (VCE - Video Coding Engine) (Value)

What is lclk? And what "values" do the lclk options take and what does it directly affect?
-u, --max-lclk=<u32>                  Maximum Data Launch Clock (Value)
-v, --min-lclk=<u32>                  Minimum Data Launch Clock (Value)

soc is a general term but what does it refer to in this case? And what does it directly affect?
-o, --max-socclk-frequency=<u32>      Maximum SoC Clock Frequency (MHz)
-p, --min-socclk-frequency=<u32>      Minimum SoC Clock Frequency (MHz)

I tried to research these things on my own but came up without proper answers to my questions.

Benman2785, 2021-01-08:
@TechieAndroid 
VCE is AMDs VideoCodingEngine - it handels AMD AMF - if you OC it you will have a higher performance on using ReLive or OBS with AMD AMF Encoder - you maybe can reduce "encoding lag" when streaming high-quality stuff // but i would not recommend that (as i dont know what max frequency would be 24/7 safe...)

lclk = fixed frequency at which the IO Hub Controller (PCIe etc) operates // have in mind that higher PCIe clocks (100MHz standard; 102 often max) will cause instability and often crash - usually the benefit is minimal

socclk-frequency = should be internal bus clock; set it to high and your PC will crash; actually i would also leave that at default

so in conclusion: if you use a Ryzen 4C/8T APU in a Laptop with a dGPU AND you want to stream with it then you could do a slight OC of VCE and increase performance of AMD AMF Encoder on he iGPU of APU and play the game only on dGPU
again, as i said, i dont know what is safe to set here = we need more info ;)

the other 2 should be left stock...

TechieAndroid, 2021-01-08:
Thanks for the feedback.

I knew about the encoder and how it works, but I didn't know what values I could set in the parameter. I'll experiment starting around 1000MHz to see what happens, and inc/dec depending on the results.

Falcosc, 2021-03-02:
Hi @TechieAndroid 
do you have an AMD Ryzen 7 3700U? If yes I would like to invite you on the PTable tests. 

My Renoir doesn't support any clock related parameters at all, so I couldn't implement table values for clock related parameters. For example: I have 3 or more clock values and don't know which of them is the min and which is max, because clock value changes are not supported on Renoir.

I hope you have windows because the Linux build isn't ready.
https://github.com/FlyGoat/RyzenAdj/issues/94

TechieAndroid, 2021-03-02:
> Hi @TechieAndroid
> do you have an AMD Ryzen 7 3700U? If yes I would like to invite you on the PTable tests.
>
> My Renoir doesn't support any clock related parameters at all, so I couldn't implement [...]

I do have a Ryzen 7 3700U (Picasso).
I also have a Ryzen 3 2200G (Raven Ridge).

I took a look at your fork but I don't really understand what's different about it from FlyGoat's RyzenAdj.

But I wouldn't mind helping out. I'd just like to know what it's for.

Falcosc, 2021-03-02:
At first only output of the power metrics to help understand the value changes better.

For example most people can not differentiate between:
- limit change message response is ok but limit is not applied (ptable reports old value): 
solution - response ok does not mean that your device does support this change, try something else to improve performance
- limit change message response is ok, limit got applied, but another limit (visible in ptable) is holding you back: 
solution - change the other limit
- limit change message response is ok, limit got applied, but a hidden limit is holding you back (ptable does report that all values are below limit)
solution - you can stop searching for improvements over SMU commands. Maybe ACPI/EC is pulling the limit

TechieAndroid, 2021-03-03:
> At first only output of the power metrics to help understand the value changes better.
>
> For example most people can not differentiate between:
> - limit change message response is ok but limit is not [...]

Okay correct me if I'm wrong but waht you're trying to say is you just want to make it so you can pull the original clock states and other values to compare before setting new ones?

Sorry the english was really broken so I had trouble reading it.

Falcosc, 2021-03-03:
I want both cases. I want the old values for the --info options for people who are unsure which values need to put in to increase a value.

And I guess more important for avoid confusion, I want the value after the change.
Because ryzenAdj change message tells that response is ok, but in reality ptable tells us that your value was not applied.