FlyGoat/RyzenAdj #80: min/max-gfxclk not working on Ryzen 5 2500U on AN515-42

Benman2785, 2021-01-07:
hey, we could fix that my AN515-42 uses fixed clocks instead of dynamic RAM speeds with the min/max-flck-frequency

now i have 2 more problems:

1. my AN515-42 only boosts from 1600MHz on Battery when i use Helle53s "RyzenTest" - than my 2500U behaves like intended/stock with 15W tdp (i guess this is either Acer with a bios bug or maybe they call it a power-saving-feature) // would be great if that could also be fixed (but is on low priority)

2. min and max-gfxclk not working on my AN515-42 with Ryzen 5 2500U
it says everything is set - but it doesnt change min or max frequency
like when i set min = 1033 and max = 1100 i get from 200MHz - 1100MHz
or when i set min = 1033 and max = 1200 i also get 200MHz - 1100MHz

here a sample picture
[image]

yacc143, 2021-01-22:
Not working on my Lenovo T14s with a Ryzen 7 Pro 4750U either:
`[root@idefix build]# ./ryzenadj --min-gfxclk=1002 --max-gfxclk=2200 --tctl-temp=100
Detected Renoir
Sucessfully set tctl_temp to 100
Failed to setmax_gfxclk_freq 
Failed to setmin_gfxclk_freq 
[root@idefix build]# cat /proc/cpuinfo | grep model.name | sort -u 
model name	: AMD Ryzen 7 PRO 4750U with Radeon Graphics
`

thieringgergo, 2021-02-10:
Not working on my T14s either, only the CPU related things seem to work on Renoir:
```
root@t14s:/home/thiering# ryzenadj --stapm-limit=6000 --fast-limit=6000 --slow-limit=6000
Detected Renoir
Sucessfully set stapm_limit to 6000
Sucessfully set fast_limit to 6000
Sucessfully set slow_limit to 6000
root@t14s:/home/thiering# ryzenadj --max-gfxclk=1200
Detected Renoir
Failed to setmax_gfxclk_freq
```

Benman2785, 2021-02-10:
i can set the clock successfully - but it doesnt use the new limit

Falcosc, 2021-02-23:
Renoir will get an unsupported error message for clock related option because we don't have message Ids for these commands and know for sure that the message Id did change compared to Picasso.

Benman2785, 2021-02-27:
will do it tomorrow

Benman2785, 2021-02-27:
@Falcosc 
RMT shows most values as 0.00000 in monitoring...

Falcosc, 2021-03-09:
Thank you for helping to clarify that min/max-gfxclk does really not have any effect on Raven.
I did add your finding in our new support matrix
https://github.com/FlyGoat/RyzenAdj/wiki/Supported-Models

Feel free to add additional findings there.

closed 2021-03-10 by FlyGoat

Falcosc, 2021-04-03:
@Benman2785 I have some news for you.

One guy managed to get them woring in his device.

He did only use 
`--min-gfxclk`

And he did check the function with msi afterburner without putting any GPU load on it.

The trick was:
Don't use multiplies of 10: 900, 910, 920, etc will not work.
But if you enter 917, it will work.

Let me know if this trick does work on your device, too. If yes, I will add it to the supported models page.

PR #113 (2021-03-10): Document which Processors are supported and which commands do work
New Support Matrix in https://github.com/FlyGoat/RyzenAdj/wiki/Supported-Models

Resolves #28
Resolves #30
Resolves #33
Resolves #54
Resolves #80
commit 04aa9560 (2021-03-10): Document which Processors are supported
Resolves #28
Resolves #30
Resolves #33
Resolves #54
Resolves #80