FlyGoat/RyzenAdj #82: Add CPU Skin Temperature Limit for

Falcosc, 2021-02-17:
Hello, 

I would like to set CPU Skin Temperature Limit.

### Why do we need it?
PPT LIMIT APU (STAPM Limit Offset 24) gets applied if SST VALUE APU (STAPM Limit Offset 92) does exceed SST LIMIT APU (STAPM Limit Offset 88, 0x38 on Renoir)
Skin temperature name in HWiNFRO is "CPU Skin Temperature" and gets set to 45° on my notebook. I need 50°C for long workloads.

### What can be done?

To fix this we need to change SST LIMIT APU. For higher accepted temperatures we could use higher PPT LIMIT APU, but I did find a way to change this Power limit, which is not importand at all, it would we enoght to add SST LIMIT APU.

To Change Skin Temperature Limit, on Renoir ec address 0x38 will accept an UInt. But value in °C need to be multiplied by 256 if you write in the same 0x38 address. Looks like the write address is maybe 8 bits off 0x38.

### Should I create a pullrequest?

I didn't prepair a Developer Environment for this project, so I could create a branch without building it. Would you like to have an untested branch?

Kind Regards
Falco

FlyGoat, 2021-02-17:
Github actions will automatically build and create binary for test.
Please create a PR to trigger Github action.

Thanks.

Falcosc, 2021-02-17:
> 
> 
> Github actions will automatically build and create binary for test.
> Please create a PR to trigger Github action.
> 
> Thanks.

A binary for each open PR? That's great, glad that I did mention it instead of starting setting up an environment.

I will keep the 256 multiplication, because I don't know anything about writing into IO registers :)

Falcosc, 2021-02-17:
Values in the lower 8 bits does actually get applied as fraction. Didn't expect this to happen. Is there a datatype which does store the lower 8 bits as a fraction? It could be that this value is actually just divided by 256.

PR #83 (2021-02-21): Add skin temperature limit for Renoir and some small improvements
Resolves #82
commit 846a43db (2021-02-21): add dGPU skin temp and fix APU skin temp value
update readme
resolves #82
closed 2021-02-21 by FlyGoat