FlyGoat/RyzenAdj #90: What should comming next? Help offered

Falcosc, 2021-02-24:
I like tinkering with my devices, so I put a lot of effort into small things. 

I love this tool so much that I decided to help all involved people by
- adding new parameters
- testing and fixing of existing ones
- clean up issues and by fixing them with a https://github.com/FlyGoat/RyzenAdj/wiki/FAQ
- extended testing to reverse engineer the effects of paramters
- creating a tuning guide to document my tinkering results https://github.com/FlyGoat/RyzenAdj/wiki/Renoir-Tuning-Guide

But there is more!

So there are some micro tweaks and knowledge to share carried over from my battery life tuning results of the worlds lightest laptop with 72Wh and adapted to my less impresive Renoir powered business laptop.

On the other hand I did understand how to access the PTable, so instead of sharing my Windows automations of RyzenAdj, I could develop the PTable integration for readout of current limits.

### Topics to order
- PTable
- Windows automation
- Battery life

### Benefits of PTable support
- did not find a linux tool to read all value mentioned here: https://github.com/FlyGoat/RyzenAdj/wiki/Renoir-Tuning-Guide
- no benefit for windows users because existing monitoring tools are much more powerful and RyzenAdj will not focus on limit monitoring
- ability to view current limits, I could resolve https://github.com/FlyGoat/RyzenAdj/issues/57
- later this could be extended to check if your value change did work (is not really useful because it does not tell if your value gets overwritten afterwards and it does not tell if other limits are holding you back)
- compact output of limit values: no need to search in a large table of values, it does only output limits which we consider as important (for example only values which have adjustment support for your platform)
- helper for all people who likes to reset there values without reboot

### Windows automation
- why I do use HWiNFO instead of Ryzen Control
- why I can't use [Mobile Ryzen Tuning](https://github.com/sbski/Renoir-Mobile-Tuning) either
- hidden tricks of HWiNFO
- benefits of windows tasks feature
- resolves https://github.com/FlyGoat/RyzenAdj/issues/68

### Battery Life
- how to get most efficient CPU peformance (high demanding workloads on battery)
- micro tweak tipps for users with battery life of 8h+ (very low power changes do only effect low power users)
- fan controll for devices which seam to not support fan controll

Topic selection was inspired of: @dcarusil @Benman2785 @kerneltoast @lazerl0rd @mthw0 @sticsk @tobiasjakobi @zhangyiming748 @mirh @storm1er 

Please tell me how you would like to order these 3 topics: PTable, Windows automation, Battery life.

Depending on the amount of responses and your prefered order I do decide to priorize a topic, skip topics or don't help at all.

FlyGoat, 2021-02-24:
I was looking for implement PMtable support myself but found it seems very hard to acces physical address directly on Windows.

Another thing comes in my mind is MSR based P-State adjust support.
Should be quiet simple.

FlyGoat, 2021-02-24:
Thanks a lot for your help!
Quiet impressive work!
As we had accumulate a lot of features and fixes on master branch, I'll make a new release if you don't have any last minute idea.

FlyGoat, 2021-02-24:
Could you please share me your IM like telegram or discord?

sticsk, 2021-02-24:
If you need help on testing stuff on Linux, I can help with whatever you want. Battery life would be nice, I also use RyzenAdj alongside with slimbookbattery [https://aur.archlinux.org/packages/slimbookbattery/](url), and when on battery saving, cpu goes down to 2.1 if I recall correctly, saving a lot of battery.

Falcosc, 2021-02-24:
@FlyGoat yes, don't have any last minute topics. I will send some message based communication possibilities via mail.
I have to thank you for starting this project :)

FlyGoat, 2021-02-24:
> @FlyGoat yes, don't have any last minute topics. I will send some message based communication possibilities via mail.
> I have to thank you for starting this project :)

Fine :-)

Falcosc, 2021-02-24:
@sticsk I do have battery life tuning stories which go beyond the obvious metrics. Much more topics about finding the break even point of the silicon. 

That's much more fun than just setting a power limit. And of course all the small things like power usage of active notebook fans and NVMe SSD differences in idle. 

To be honest, some of them are pointless for RyzenAdj audience because Ryzen isn't as good as intel in idle. But there is a small amount of real world use cases which have phases where idle actually matters.

tobiasjakobi, 2021-03-05:
I don't use Windows, so I guess I'm mostly interested in the PTable support. Anyway, it's nice to see that people are actively working on this, keep it up! :-)

Falcosc, 2021-03-24:
@tobiasjakobi @sticsk 
I did blindly implement physical memory mapping for Linux to support ptable.

You could give it a try, but it would surprise me if it does actually work.

The branch name is ptable https://github.com/FlyGoat/RyzenAdj/tree/ptable

tobiasjakobi, 2021-03-25:
@Falcosc Hey, thanks for the heads up.

I checked the branch, but as you already guessed, it doesn't really work.

```
> ryzenadj --info
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
CPU Family: Renoir
SMU_SERVICE REQ_ID:0x3
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x11, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU BIOS Interface Version: 17
Version: v0.8.0 
SMU_SERVICE REQ_ID:0x6
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x370003, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
PM Table Version: 370003
SMU_SERVICE REQ_ID:0x66
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0xcbeda000, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x65
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
|       Name       |   Value   |      Paramter      |
|------------------|-----------|--------------------|
| STAPM LIMIT      |    50.000 | stapm-limit        |
| STAPM VALUE      |     0.000 |                    |
| PPT LIMIT FAST   |     0.000 | fast-limit         |
| PPT VALUE FAST   |     0.000 |                    |
| PPT LIMIT SLOW   |     0.000 | slow-limit         |
| PPT VALUE SLOW   |     0.000 |                    |
| StapmTimeConst   |     0.000 | stapm-time         |
| SlowPPTTimeConst |     0.000 | slow-time          |
| PPT LIMIT APU    |     0.000 | apu-slow-limit     |
| PPT VALUE APU    |     0.000 |                    |
| TDC LIMIT VDD    |     0.000 | vrm-current        |
| TDC VALUE VDD    |     0.000 |                    |
| TDC LIMIT SOC    |     0.000 | vrmsoc-current     |
| TDC VALUE SOC    |     0.000 |                    |
| EDC LIMIT VDD    |     0.000 | vrmmax-current     |
| EDC VALUE VDD    |     0.000 |                    |
| EDC LIMIT SOC    |     0.000 | vrmsocmax-current  |
| EDC VALUE SOC    |     0.000 |                    |
| THM LIMIT CORE   |     0.000 | tctl-temp          |
| THM VALUE CORE   |     0.000 |                    |
| STT LIMIT APU    |     0.000 | apu-skin-temp      |
| STT VALUE APU    |     0.000 |                    |
| STT LIMIT dGPU   |     0.000 | dgpu-skin-temp     |
| STT VALUE dGPU   |     0.000 |                    |
```

```
> ryzenadj --dump-table
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x66
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0xcbeda000, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x65
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xfd, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
Transfer table was rejected
SMU_SERVICE REQ_ID:0x65
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xfd, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
Transfer table was rejected
SMU_SERVICE REQ_ID:0x6
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x370003, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x65
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xfd, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
Transfer table was rejected
PM Table Dump of Version: 370003
| Offset |    Data    |   Value   |
|--------|------------|-----------|
| 0x0000 | 0x00000000 |     0.000 |
<snip>
| 0x08A8 | 0x00000000 |     0.000 |
```

Falcosc, 2021-03-25:
@tobiasjakobi Yes and no. At least we got one number.

I did try to workarround the linux issues with transfer table. Please have a look on the latest version
https://github.com/FlyGoat/RyzenAdj/archive/refs/heads/ptable.tar.gz

tobiasjakobi, 2021-03-25:
Yes, that looks better.

I've put the output in a separate Gist because of the length:
https://gist.github.com/tobiasjakobi/c45d15024feb6af346e2d9485d06af67

Falcosc, 2021-03-25:
Great, only a single `REQ_ID:0x65` have a close look at these because I don't know how effective my workaround will be.

tobiasjakobi, 2021-03-26:
Thanks, you let me know if something else I can test.

Anyway, this retry approach looks similar to how e.g. drmIoctl() [reference](https://gitlab.freedesktop.org/mesa/drm/-/blob/master/xf86drm.c#L202) works.