FlyGoat/RyzenAdj #91: Does anyone need a Low Power Profile for Cable or a High Power Profile for Battery?

Falcosc, 2021-02-24:
Hi Folks,

I just found out what these AC/DC Power source commands are doing.
There are two of them, one for Power over Cable `0x11` and one for Power over Battery `0x12`

These get automatically applied on plug or unplug charging cable.

I figured out what these things do change. Both can be applied on cable and both can be applied on battery. So you can switch to high performance on battery or switch to maximum power savings while on cable.

0x11 Cable Mode
- change all CLK FIT Values to 50
- Infinity Fabric is nearly pinned, even in Idle with power saving profile It down clocks only to a minimum of MAX-80Mhz
 
0x12 Battery Mode
- change all CLK FIT Values to 95
- Minimum Infinity Farbic speed of 400MHz at idle but still boosting to max

Don't know what these FIT things are doing, maybe they control the power saving during idle.

Things we all know is the importance of the Infinity Fabric, but what I didn't know is the high power draw.

At Idle this downclock to 400Mhz does save 50% Package Power. It goes from 4W to 2W.

But here is the question for you, Could anybody request something like:
- I want to change the word, give me these 2W of energy saving while on Cable
- Even on battery I want maximum snappiness, High Performance Power Plan is not enough! Don't care if the battery lasts only 3,5 instead of 4 hours

If yes, then we could add these weird messages to ryzenAdj. But I don't want to add unnecessary commands to avoid even more confusion.

Falcosc, 2021-03-09:
@storm1er Could I raise this question in your Repo as well? It is difficult to ask people things if they just visit the issue overview if they want to complain about something ;)

storm1er, 2021-03-29:
I think this is more a question for @FlyGoat than me.

This could be useful to match settings applied on triggers "battery plugged in/disconnected".

Falcosc, 2021-03-29:
It's a question to the users :)

It's is not neccesary to match it with the power source, because devices does already use this modes and set it accordingly on power source change, thats how I did discover them.

It is only usefull if you want to unmatch the setting: AC Mode for Battery or Batter Mode for Cable.

storm1er, 2021-03-29:
I think the problem is more about how users want to use the software than how we want them to use it.  
People want to get the most power out of their laptop.

So the idea would be to always set "full power mode" by default and add an option to be able to do the opposite whatever the laptop state, battery or cable.

storm1er, 2021-03-29:
> @storm1er Could I raise this question in your Repo as well? It is difficult to ask people things if they just visit the issue overview if they want to complain about something.

Yes you can :)

Falcosc, 2021-03-29:
Understood, so you just mentioned that the default value in an UI integration should match the power source.

storm1er, 2021-03-29:
Using ryzenadj always set full power mode and adding an argument `--battery-mode` set the battery mode.

In the UI, I will add a checkbox "battery mode" =)

Falcosc, 2021-03-29:
We don't know which mode is currently selected, so would be not good to send AC Mode command to SMU if no options are given.
We may need 2 flags because I don't like option string parsing.

And there are people out there who use ryzenAdj to save power because other than Intel it is currently not possible to limit the clock speed with windows power plan. Both hidden options for CPU speed doesn't work if you enable them in the power plan layout.

storm1er, 2021-03-30:
> We don't know which mode is currently selected, so would be not good to send AC Mode command to SMU if no options are given.

Right

> Currently, ryzenAdj doesn't send any commands if no options are given. And we would like to keep this behavior to only send commands if options are set.

To keep coherence, I agree with that

> We may need 2 flags because I don't like option string parsing.

I agree, but it means we must ensure good behavior when both options are set.

> And there are people out there who use ryzenAdj to save power
 
Good point

Falcosc, 2021-03-30:
I would keep it simple, just let them overwrite each other.

Like stamp-limt on Rainor just gets overwritten by fast-limit or slow-limit depending on the average power usage during the last slow-time seconds.

On Reddit somebody did tell my about 10s boost delay on battery. Today I did confirm this problem and even better, the reason for that is the setpoint option which can currently only be controlled by the battery or cable mode. So now it is clear that this would actually make real performance changes. One line below the setpoint values in ptable are the current values. So, you can monitor when the usage number goes above the setpoint to trigger boost. It does actually take 9-10s depending on how low you did start in idle.

But at first, I would like to wait for processing the existing pull requests. Didn't hear from flygoat since nearly a week.

PR #129 (2021-04-04): Add power-saving and max-performance settings
to control hidden options like boost delay on battery
Resolves #91
commit 58476d8c (2021-04-04): Add power-saving and max-performance settings
to control hidden options like boost delay on battery
Resolves #91
closed 2021-04-04 by FlyGoat