FlyGoat/RyzenAdj #94: Raven and Picasso Testers for Windows needed

Falcosc, 2021-02-27:
Hi,

I'm about to introduce PTable readouts, but I need some of you to help me finding the positions of all the Clock related values.
I already did add all Renoir related lines. Now only Raven and Picasso is missing.

For the following options we need help. I want to know at which offset these are located in your Ptable. 
```
--max-gfxclk-freq
--min-gfxclk-freq
--max-socclk-freq
--min-socclk-freq
--max-fclk-freq
--min-fclk-freq
--max-vcn
--min-vcn
--max-lclk
--min-lclk
```

To test them I did create a Debug Build for you.
https://github.com/Falcosc/RyzenAdj/runs/1995149004 Download link is top right at Artefacts [ryzenadj-win64](https://github.com/Falcosc/RyzenAdj/suites/2139485195/artifacts/43757354)

Don't forget all libraries from the prebuilt, a new one was added to support physical memory access.

It is very easy to test. Just use the `--dump-table` option. It will print the values before adjustments and the values after the adjustments. So it is very easy to find them.

You only need to make sure to select values which are valid to keep them correctly applied.

For example if I would search for the `--stapm-limit` I would select a unique value like 27500, but don't 27123 because I am not sure if it could apply the 123mA.

`ryzenadj.exe --dump-table --stapm-limit=27000 2>nul`
```
Sucessfully set stapm_limit to 27000
PM Table Dump of Version: 370003
| Offset |    Data    |   Value   | After Adjust |
|--------|------------|-----------|--------------|
| 0x0000 | 0x420C0000 |    35.000 |       27.000 |
| 0x0004 | 0x40810CE6 |     4.033 |        4.149 |
| 0x0008 | 0x420C0000 |    35.000 |       35.000 |
| 0x000C | 0x4093C14C |     4.617 |        9.740 |
| 0x0010 | 0x41A80001 |    21.000 |       21.000 |
| 0x0014 | 0x40969ED2 |     4.707 |        4.707 |
```

This does tell us that `--stapm-limit` is located at offset 0x0000 for the Table Version `370003`

Could you please help me and try to find these?
```
--max-gfxclk-freq
--min-gfxclk-freq
--max-socclk-freq
--min-socclk-freq
--max-fclk-freq
--min-fclk-freq
--max-vcn
--min-vcn
--max-lclk
--min-lclk
```
You need a Raven or Picasso APU to which does support these parameters.

It is expected that some of you could get a popup that something is broken. Because this is a debug build which does complain about issues. Please share them to be able to improve it, I did already found an old one https://github.com/FlyGoat/RyzenAdj/pull/92
If you get don't get the table and the error messages are not enough, please remove the `2>nul` part of the command line to share all the SMU request information.

Thank you!

Benman2785, 2021-02-27:
will do it tomorrow for my Raven Ridge 2500U

Benman2785, 2021-02-27:
@Falcosc 

i get vcruntime errors...

2010, 2012, 2013, 2015-19 installed...

so i cant help you as of now

Benman2785, 2021-02-27:
vcruntime140d.dll and ucrtbased.dll
both are installed and no other programm have issues

Falcosc, 2021-02-28:
Ok, I guess Debug Builds need more preconditions.

What about an release Build?
https://github.com/Falcosc/RyzenAdj/suites/2141571873/artifacts/43809443 from https://github.com/Falcosc/RyzenAdj/pull/2/checks?check_run_id=1997206549

Benman2785, 2021-02-28:
will test it later today

Benman2785, 2021-02-28:
i can say it doesnt work - only ONE value in monitoring isnt 0.0000

rename to .bat instead of .bat.txt
[dump.bat.txt](https://github.com/FlyGoat/RyzenAdj/files/6057185/dump.bat.txt)

results in:
[info.txt](https://github.com/FlyGoat/RyzenAdj/files/6057192/info.txt)

Falcosc, 2021-02-28:
Thank you very much. But without the debug build we can not see if the memory address makes sense.

I hope that your get memory address for table command is just returning a wrong address. Because if I extract values from memory position 0 it will only collect garbage.

I will create a release build with debug messages for you :)

Benman2785, 2021-02-28:
ok ;)

Benman2785, 2021-02-28:
@Falcosc 
if you can do it the next 3h i can still test it today ;)

Falcosc, 2021-02-28:
Sorry, did took a bit longer then expected because I did need to fix a memory issue. 

https://github.com/Falcosc/RyzenAdj/suites/2143516502/artifacts/43850612

Should look like this now:
```
ryzenadj.exe --dump-table -a 28000
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x11, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x6
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x370003, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x66
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0xca571000, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x65
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x65
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x14
SMU_SERVICE REQ: arg0: 0x6d60, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x6d60, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
Sucessfully set stapm_limit to 28000
PM Table Dump of Version: 370003
SMU_SERVICE REQ_ID:0x65
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
| Offset |    Data    |   Value   | After Adjust |
|--------|------------|-----------|--------------|
| 0x0000 | 0x420C0000 |    35.000 |       28.000 |
| 0x0004 | 0x409CFD46 |     4.906 |        4.980 |
| 0x0008 | 0x420C0000 |    35.000 |       35.000 |
```

Benman2785, 2021-02-28:
i will test in 90min - but dont have much time today ;)

Falcosc, 2021-02-28:
Based on the result you will get a short answer about missing support or a long answer with description how to find the command to get the dram address.

Benman2785, 2021-02-28:
`RyzenAdj dumps
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x6, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
[... 125 lines cut ...]
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
[... cut ...]
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0`

Benman2785, 2021-02-28:
> RyzenAdj dumps
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x6, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
[... 127 lines cut ...]
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
[... cut ...] rg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0

>

Benman2785, 2021-02-28:
> RyzenAdj dumps
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x1
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x6, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x1a
SMU_SERVICE REQ: arg0: 0x6d60, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x6d60, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
Sucessfully set stapm_limit to 28000
PM Table Dump of Version: 1e0004
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0xff, arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
| Offset |    Data    |   Value   | After Adjust |
|--------|------------|-----------|--------------|
| 0x0000 | 0x00000101 |     0.000 |        0.000 |
| 0x0004 | 0x00000100 |     0.000 |        0.000 |
| 0x0008 | 0x00000000 |     0.000 |        0.000 |
| 0x000C | 0x00000000 |     0.000 |        0.000 |
| 0x0010 | 0x00000000 |     0.000 |        0.000 |
| 0x0014 | 0x00000000 |     0.000 |        0.000 |
| 0x0018 | 0x00000000 |     0.000 |        0.000 |
| 0x001C | 0x00000000 |     0.000 |        0.000 |
| 0x0020 | 0x00000000 |     0.000 |        0.000 |
| 0x0024 | 0x00000000 |     0.000 |        0.000 |
| 0x0028 | 0x00000000 |     0.000 |        0.000 |
| 0x002C | 0x00000000 |     0.000 |        0.000 |
| 0x0030 | 0x00000000 |     0.000 |        0.000 |
| 0x0034 | 0x00000000 |     0.000 |        0.000 |
| 0x0038 | 0x00000000 |     0.000 |        0.000 |
| 0x003C | 0x00000000 |     0.000 |        0.000 |
| 0x0040 | 0x00000000 |     0.000 |        0.000 |
| 0x0044 | 0x00000000 |     0.000 |        0.000 |
| 0x0048 | 0x00000000 |     0.000 |        0.000 |
| 0x004C | 0x00000000 |     0.000 |        0.000 |
| 0x0050 | 0x00000000 |     0.000 |        0.000 |
| 0x0054 | 0x00000000 |     0.000 |        0.000 |
| 0x0058 | 0x00000000 |     0.000 |        0.000 |
| 0x005C | 0x00000000 |     0.000 |        0.000 |
| 0x0060 | 0x00000000 |     0.000 |        0.000 |
| 0x0064 | 0x00000000 |     0.000 |        0.000 |
[... 378 lines cut ...]
| 0x0650 | 0x00000000 |     0.000 |        0.000 |
| 0x0654 | 0x00000000 |     0.000 |        0.000 |
| 0x0658 | 0x00000000 |     0.000 |        0.000 |
| 0x065C | 0x00000000 |     0.000 |        0.000 |
| 0x0660 | 0x00000000 |     0.000 |        0.000 |
| 0x0664 | 0x00000000 |     0.000 |        0.000 |
| 0x0668 | 0x00000000 |     0.000 |        0.000 |
| 0x066C | 0x00000000 |     0.000 |        0.000 |
| 0x0670 | 0x00000000 |     0.000 |        0.000 |
| 0x0674 | 0x00000000 |     0.000 |        0.000 |
| 0x0678 | 0x00000000 |     0.000 |        0.000 |
| 0x067C | 0x00000000 |     0.000 |        0.000 |
| 0x0680 | 0x00000000 |     0.000 |        0.000 |
| 0x0684 | 0x00000000 |     0.000 |        0.000 |
| 0x0688 | 0x00000000 |     0.000 |        0.000 |
| 0x068C | 0x00000000 |     0.000 |        0.000 |
| 0x0690 | 0x00000000 |     0.000 |        0.000 |
| 0x0694 | 0x00000000 |     0.000 |        0.000 |
| 0x0698 | 0x00000000 |     0.000 |        0.000 |
| 0x069C | 0x00000000 |     0.000 |        0.000 |
| 0x06A0 | 0x00000000 |     0.000 |        0.000 |
| 0x06A4 | 0x00000000 |     0.000 |        0.000 |
| 0x06A8 | 0x00000000 |     0.000 |        0.000 |

>

Falcosc, 2021-02-28:
Ok, that is clear

`0xB` is not your dram address because it did return `0`

`0xC` is correct because `0x1e0004` is a known table version

But real bad is `0x3D` because this is also wrong. It should be your transfer Table to DRAM message. Fixing `0xB`  is easy because we just need to find a command which does respond with a large number.
But finding your transfer table command could be a nightmare. You would need to dump table after each command you did try and look if any value did change.

Please try to find your get dram adress command with https://github.com/irusanov/SMUDebugTool
It has also a scan for Ptable button, not sure how good this button is. But if it is really guessing, then it could take a while to execute.

These are your SMU adresses: 
```
smu->msg = 0x3B10524;
smu->rep = 0x3B10570;
smu->arg_base = 0x3B10A40;
```

Benman2785, 2021-02-28:
[info2.txt](https://github.com/FlyGoat/RyzenAdj/files/6057908/info2.txt)
here is the list edited to each command - but its not shortened

btw - how to shorten? only post comment with correct adress?

please tell me what to do with SMUDebugTool 
and today i dont have much time left - need to shower because i get up early for work next day

Falcosc, 2021-02-28:
With shorten I mean the large comments which forces anybody to scroll. Text files or normal size comments are ok.

SMUDebugTool
You use the SMU tab, put the 3 addresses in
And then you can test messages from `0x1` up to whatever you want. But everything should be below `0x40`

Use `0x0` as argument for the message we are looking for.

SMU or APU could get damaged if wrong values are transmitted. So don't try high values.

Benman2785, 2021-02-28:
> CMD:  0x03B10528
RSP:  0x03B10564
ARG:  0x03B10998

CMD:  0x03B10A20
RSP:  0x03B10A80
ARG:  0x03B10A88

CMD:  0x03B10A24
RSP:  0x03B10A84
ARG:  0x03B10A8C

>

Falcosc, 2021-02-28:
```
msg = 0x3B10524;
rep = 0x3B10570;
arg = 0x3B10A40;
command id= 0x1 - 0x80
argument = 0x0
```

Benman2785, 2021-02-28:
i have a ptable scan for you from last - but i cant copy or export it from powertable

Falcosc, 2021-02-28:
Do you have commands and addresses from the ptable scan, or did it just open your table? If it did open it, does it have meaningfull values?

Benman2785, 2021-02-28:
0x0000 28.000000
0x0004 4.902752
0x0008 30.00000
0x000C 4.3328...
0x0010 25.00...

etc

Benman2785, 2021-02-28:
so yes it has values that make sense

Falcosc, 2021-02-28:
ok I can get the adresses from zenstate-core (already did it, but I must made a misstake)

Maybe the ussed command Ids are logged in the little box.

Could you clear the box and hit the ptable button again please?

Benman2785, 2021-02-28:
there are roughly 200-250 values that are not 0

Benman2785, 2021-02-28:
clear what box?
and only 15min left

Falcosc, 2021-02-28:
It does log the messages. Close and reopen the smu debug tool

Benman2785, 2021-02-28:
[image]
default values give me:

CMD:  0x03B10528
RSP:  0x03B10564
ARG:  0x03B10998

CMD:  0x03B10A20
RSP:  0x03B10A80
ARG:  0x03B10A88

CMD:  0x03B10A24
RSP:  0x03B10A84
ARG:  0x03B10A8C

Falcosc, 2021-02-28:
Ok, I have to manually find it. At least I know that there are known by this tool. I am just not smart enough to find out why the values which I took from there are the wrong ones.

Benman2785, 2021-02-28:
i can help tomorrow - in 20h
but today i dont have more time.

just tell me what and how to do and i will do it ;)

Falcosc, 2021-02-28:
Maybe I have a new version until then :)

Benman2785, 2021-02-28:
:)

Falcosc, 2021-02-28:
Ok found our problem. The arguments are very different on your APU

https://github.com/irusanov/SMUDebugTool/blob/908a4465d68c171f85ec9170a24978914b392686/Utils/Ops.cs#L478

Thanks for testing, wouldn't have found this problem without you.

Benman2785, 2021-03-01:
glad i could help ;)

any more testing needed?

Falcosc, 2021-03-01:
Yes

Falcosc, 2021-03-01:
Could you please test some messages with SMU debug tool?

All test commands use these values:
```
MSG = 0x3B10524;
RSP = 0x3B10570;
ARG = 0x3B10A40;
```
Test 1:
```
command id= 0xB;
arguments = 0x3;
Send and write down response

command id= 0xB;
arguments = 0x5;
Send and write down response
```

Benman2785, 2021-03-01:
will do it tomorrow - no time today (first day in office is always hard) ;)

Falcosc, 2021-03-02:
Here is the new release build https://github.com/Falcosc/RyzenAdj/suites/2161301435/artifacts/44339060 from here https://github.com/Falcosc/RyzenAdj/runs/2015511015

Benman2785, 2021-03-02:
hi,

will test it tomorrow - had a horror trip back from work xD (needed 90mins longer than expected - screaming children everywhere)

sry that you have to wait

Benman2785, 2021-03-03:
command id= 0xB;
arguments = 0x3;

= PC freezes!

Benman2785, 2021-03-03:
command id= 0xB;
arguments = 0x5;

= PC freezes after 1-3s

Benman2785, 2021-03-03:
command id= 0xA;
arguments = 0x3;

= Bluescreen "thread stuck in device driver"

Benman2785, 2021-03-03:
command id= 0xB;
arguments = 0x3;

= "unknows status"
no other result - no output in SMUDebug

Benman2785, 2021-03-03:
command id= 0xA;
arguments = 0x5;

= Bluescreen "thread stuck in device driver" - before that it is VERY unresponsive - mouse "hangs" for some seconds

Benman2785, 2021-03-03:
command id= 0xB;
arguments = 0x5;

= "unknows status"
no other result - no output in SMUDebug

Benman2785, 2021-03-03:
[info.txt](https://github.com/FlyGoat/RyzenAdj/files/6079148/info.txt)
[info2.txt](https://github.com/FlyGoat/RyzenAdj/files/6079154/info2.txt)

this is result with new built
it adds some different memory addresses

Benman2785, 2021-03-03:
MSG = 0x3B10528;
RSP = 0x3B10564;
ARG = 0x3B10998;
(SMU default)

results:
cid= 0xA
arg=0x3

HEX: 0x2, 0x0, 0x0, 0x0, 0x0, 0x0
INT: 2, 0, 0, 0, 0, 0

arg=0x5
HEX: 0x2, 0x0, 0x0, 0x0, 0x0, 0x0
INT: 2, 0, 0, 0, 0, 0

___
cid= 0xB
arg=0x3
HEX: 0x32820000, 0xB5, 0x0, 0x0, 0x0, 0x0
INT: 847380480, 181, 0, 0, 0, 0

arg=0x5
HEX: 0xDF00, 0xD9E200, 0x0, 0x0, 0x0, 0x0
INT: 57088, 14279168, 0, 0, 0, 0

Falcosc, 2021-03-03:
Thank you so much. But how did you get the last response? 
```
cid= 0xB
arg=0x3
HEX: 0x32820000, 0xB5, 0x0, 0x0, 0x0, 0x0
INT: 847380480, 181, 0, 0, 0, 0

arg=0x5
HEX: 0xDF00, 0xD9E200, 0x0, 0x0, 0x0, 0x0
INT: 57088, 14279168, 0, 0, 0, 0
```

looks better then bluescreen

Benman2785, 2021-03-03:
[image]
here the table values with
MSG = 0x3B10524;
RSP = 0x3B10570;
ARG = 0x3B10A40;

Benman2785, 2021-03-03:
[image]
remaining values ;)

Benman2785, 2021-03-03:
btw - i nowhere see 1100 as value
so i assume you cant set max gpu clock with SMU - atleast in my version

Falcosc, 2021-03-03:
Yes, you did already have confirmed that ingame. Ptable is now the proove.

But FCKL should be visible.

And Sorry for giving you the wrong SMU addresses.
```
#define PSMU_C2PMSG_MESSAGE_ADDR          0x3B10a20
#define PSMU_C2PMSG_RESPONSE_ADDR         0x3B10a80
#define PSMU_C2PMSG_ARG_BASE              0x3B10a88
```

Benman2785, 2021-03-03:
should i retest again? ;)

Falcosc, 2021-03-03:
You can proof the values:
```
B + 3 = 0x6f7a8000
B+ 5 = 0x6f7a8620
```

Because I don't understand why these are looking like the start and end address of the table. But I give you a release to test both addresses tomorrow if I don't figure out the reason.

Benman2785, 2021-03-03:
ok - i will be back tomorrow around same time ;)

edit - i test thos 2 values fast

Benman2785, 2021-03-03:
3
HEX: 0x6F7A8000, 0x0, 0x0, 0x0, 0x0, 0x0
INT: 1870299136, 0, 0, 0, 0, 0
---
5
HEX: 0x6F7A8620, 0x0, 0x0, 0x0, 0x0, 0x0
INT: 1870300704, 0, 0, 0, 0, 0

Falcosc, 2021-03-03:
thank you. But don't understand how :)

Benman2785, 2021-03-03:
k

Benman2785, 2021-03-03:
3
HEX: 0x6F7A8000, 0x0, 0x0, 0x0, 0x0, 0x0
INT: 1870299136, 0, 0, 0, 0, 0
5
HEX: 0x6F7A8620, 0x0, 0x0, 0x0, 0x0, 0x0
INT: 1870300704, 0, 0, 0, 0, 0

[image]

[20 more replies]

closed 2021-03-05 by Falcosc