FlyGoat/RyzenAdj #124: Tag a release with the new versions

ericonr, 2021-03-24:
Hi!  The history of the https://github.com/FlyGoat/RyzenAdj/blob/master/lib/ryzenadj.h file shows it was changed to a 0.7.1 version and then a 0.8.0, but none of those prompted a release being tagged in the repo.

I just backported https://github.com/FlyGoat/RyzenAdj/commit/6efee1997c468150f472941f0f183fc054b436a4 into Void Linux's package, but it would be nice if there were proper releases to pull from.

Falcosc, 2021-03-24:
We skipped the 0.7.1 bug fix release because it was not worth a tag, the error was existing for years :)

Falcosc, 2021-03-24:
@ericonr here is the full 07.1 release description https://github.com/FlyGoat/RyzenAdj/releases/tag/v0.7.1

Could you tell me how dangerous this old 10 byte long allocation overflow actually is on modern operating systems? Did we really overwrite random 10 bytes of other applications or does the OS reserve some space in memory to make sure that small overflows don't hurt other applications?

ericonr, 2021-03-24:
No overflow size can hit memory used by other applications, since all modern OSs use virtual memory on x86 platforms. The issue with overflows is that they might alter other things allocated in the heap (so other variables and structs) in your own program.

I had to backport the patch because otherwise RyzenAdj made musl's allocator abort the program, since it detected that the application had written beyond the memory allocated to it (for that specific pointer). Aborting when you detect such a thing is really the only valid behavior, because a buffer overflow can also be caused by bugs in your program or especially crafted input taking advantage of such bugs, and can lead to all sorts of issues, like executing attacker controlled code or leaking secret information.

ericonr, 2021-03-24:
I'm not sure why glibc didn't catch it.

Falcosc, 2021-03-24:
So, it was the first time that you use musl's allocator for ryzenAdj? Or did you create a workaround for old releases?

I did discover it accidentally, too. The msvc debug build didn't find the issue during compile time, but it did find it during runtime.

I thought if it can be detected during runtime, maybe something can at least map virtually map the overflow to the scope of our application to only overwrite our stuff instead of overwriting other applications.

ericonr, 2021-03-24:
Look into running the program under `valgrind` later, it might help spot other such issues :)

> I guess, it was the first time that you use musl's allocator for ryzenAdj? Or did you create a workaround for old releases?

I'm not sure what happened, just had a complaint from a user, saw the commit and added it to our package. Idk if it appeared after the update or if no one had tried the program on musl yet.

Falcosc, 2021-03-24:
The cool kids don't use conservative tools with overcautiously error checks like musl.
They just believe that every program is perfect. 

Maybe the 10 bytes overflow was on purpose to try to crash the power management software which was holding you back all the time 😄

ericonr, 2021-03-24:
lol :D 

Thanks for the tag, can you comment in here when it's made a full "release" instead of only "pre-release"? Void prefers packaging only full releases :)

Falcosc, 2021-03-24:
Yes. I suggest waiting for @FlyGoat approval of my release description.

FlyGoat, 2021-03-25:
@Falcosc Yes go ahead.
I had v0.7.1 in release draft but somehow forgot to press release button :-(
Thanks.

FlyGoat, 2021-03-25:
The head overflow is always dangerous tough...
Actually ASAN could detect to but I'm just to lazy lol

Falcosc, 2021-03-25:
@ericonr button is pressed

Actually I needed to press 3 buttons, Outrage!

closed 2021-03-25 by Falcosc