FlyGoat/RyzenAdj #133: HP 14s-dk0093AU, Ryzen 3500U error running --info / --dump-table

Abh15h3k, 2021-04-06:
I am running archlinux. I have built ryzenadj from source.
most of the arguments take effect and i can notice a performance difference. but i get an error running `sudo ./ryzenadj --info` (or --dump-table).
exact output running `sudo ./ryzenadj --info`
```
CPU Family: Picasso
SMU Bios Interface Version: 19
version v0.8.0
failed to map memory
refresh_table failed
unable to init power metric table: -5
```
exact output running `sudo ./ryzenadj --stapm-limit 30000 --dump-table`
```
failed to map memory
refresh_table failed
unable to init power metric table: -5
```

Falcosc, 2021-04-06:
Finally, someone with a hardened Linux system ;)

I need your system to create a nice error message.

Falcosc, 2021-04-06:
Please create a non release build to get some debug messages, maybe your PM Table address is not aligned to page size. But I bet it's the known memory access violation issue on hardened systems, even if you have /dev/mem

Falcosc, 2021-04-06:
@Abh15h3k please use my `Falcosc:mmaperror` branch
https://github.com/FlyGoat/RyzenAdj/pull/134/files

I will add error handling there until we get a nice result for hardened Linux systems because I don't have access to an affected system.

Abh15h3k, 2021-04-06:
> Finally, someone with a hardened Linux system ;)
> 
> I need your system to create nice error messages.
> 
> You don't have /dev/mem right?

not sure what you mean be hardened linux system ( i don't remember doing any hardening ).
im using the regular linux kernel `pacman -S linux, linux-headers`.
Also `/dev/mem` is present

> If you have memory access security issue, you need to install this:
> https://gitlab.com/leogx9r/ryzen_smu

I have this kernel module install and loaded.
```
[ 7552.526556] ryzen_smu: CPUID: family 0x17, model 0x18, stepping 0x1, package 0x0
[ 7552.527278] ryzen_smu: SMU v4.30.64.0
```

> please use my Falcosc:mmaperror branch
> https://github.com/FlyGoat/RyzenAdj/pull/134/files

built with `cmake ..` and `make`

```
sudo ./ryzenadj --info

SMU_SERVICE REQ_ID:0x3
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x13, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
init_table
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x3, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0xdaadf000, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x3, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x3, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
CPU Family: Picasso
SMU BIOS Interface Version: 19
Version: v0.8.0 
failed to map memory
refresh_table failed
Unable to init power metric table: -5
```

```
sudo ./ryzenadj --stapm-limit 30000 --dump-info

init_table
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x3, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0xdaadf000, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x3, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x3, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
failed to map memory
refresh_table failed
Unable to init power metric table: -5
```

Falcosc, 2021-04-06:
@Abh15h3k thank you, your address `0xdaadf000` seems to be aligned.

Could you please use my branch or apply the diff to get the mmap error reason?
`Falcosc:mmaperror`
https://github.com/FlyGoat/RyzenAdj/pull/134/files

After we did create a nice error message for all users without memory access rights we can include support fort the kernel module `ryzen_smu`
But first lets create a nice error message, I don't want to suggest the kernel module for all mmap errors, so I would like to see what your error is.

Abh15h3k, 2021-04-06:
>Could you please use my branch or apply the diff to get the mmap error reason?
>Falcosc:mmaperror
>https://github.com/FlyGoat/RyzenAdj/pull/134/files

the outputs in the previous comment were from your build

Falcosc, 2021-04-06:
> the outputs in the previous comment were from your build

please try again, you may forgot something, even if the error string is empty, you would still see the ":"

Old version: `failed to map memory`
new version `failed to map memory: %s`

Abh15h3k, 2021-04-06:
TLDR: `failed to map memory: Operation not permitted`

oops. cloned the repo, forgot to change the branch.
```
sudo ./ryzenadj --info

SMU_SERVICE REQ_ID:0x3
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x13, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
init_table
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x3, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0xdaadf000, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x3, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x3, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
CPU Family: Picasso
SMU BIOS Interface Version: 19
Version: v0.8.0 
failed to map memory: Operation not permitted
refresh_table failed
Unable to init power metric table: -5
```

```
sudo ./ryzenadj --stapm-limit 30000 --dump-info

init_table
SMU_SERVICE REQ_ID:0xc
SMU_SERVICE REQ: arg0: 0x0, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x1e0004, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0xb
SMU_SERVICE REQ: arg0: 0x3, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0xdaadf000, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REQ_ID:0x3d
SMU_SERVICE REQ: arg0: 0x3, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
SMU_SERVICE REP: REP: 0x1, arg0: 0x3, arg1:0x0, arg2:0x0, arg3:0x0, arg4: 0x0, arg5: 0x0
failed to map memory: Operation not permitted
refresh_table failed
Unable to init power metric table: -5
```

Falcosc, 2021-04-06:
Thank you, the creator of https://gitlab.com/leogx9r/ryzen_smu told me about the hardened systems. But I couldn't find anybody who has one.

Could you check if the kernel module has access? Just to make sure ;)
Here are some examples: https://gitlab.com/leogx9r/ryzen_smu/-/tree/master/scripts

I guess you need dump table and read dump

Falcosc, 2021-04-06:
As far as I understood, giving root access to all memory areas are considered as a security issue. That's why you can have `/dev/mem`, but you are not allowed to read all memory.

Distros like Ubuntu are not so strict about that, if you are root there you can have a look into all memory ;)

Abh15h3k, 2021-04-06:
the `dump_pm_table.py` script dumps 15 instances of the table while idling and 15 instance under load (not sure what load. the script does it by itself). then i used `read_dump.py` to read one of the idle and one of the load dumps.

idle dump: https://pastebin.com/B9kmjTNn
load dump: https://pastebin.com/VDTKKz2A

also attaching all the dumps (30 in total)
[pm_dumps.zip](https://github.com/FlyGoat/RyzenAdj/files/6263713/pm_dumps.zip)

Falcosc, 2021-04-06:
Looks good, so kernel module is your solution.

Let's check if we could convince your system that mmap isn't so dangerous before implementing the kernel module.

Please replace
https://github.com/Falcosc/RyzenAdj/blob/mmaperror/lib/libpci.c#L70
`MAP_SHARED` with
- test1: MAP_PRIVATE
- test2: MAP_ANONYMOUS

Abh15h3k, 2021-04-06:
> test1: MAP_PRIVATE

`failed to map memory: Operation not permitted`
> test2: MAP_ANONYMOUS

`failed to map memory: Invalid argument`

Falcosc, 2021-04-06:
Have a look at my new error message:
https://github.com/FlyGoat/RyzenAdj/pull/134/files

If these are looking fine I will start adding the support for the mentioned kernel module 4 hours later.

Abh15h3k, 2021-04-06:
just to make sure i understand you will be working on this
`We do support usage of this kernel module https://gitlab.com/leogx9r/ryzen_smu`
right?

> I will start adding the support for the mentioned kernel module 4 hours later.

no hurry.

Falcosc, 2021-04-06:
Yes, but I can't test these things at all, so I do need you even for things like "does my error message display correctly"

Falcosc, 2021-04-06:
Yes, but I can't test this at all. I do even need you for tests like "does my error message display correctly"

After I'm done it will just use the kernel module instead of showing the error message.

Abh15h3k, 2021-04-06:
for `MAP_SHARED` and `MAP_PRIVATE` your error message is displayed. along with `failed to map memory: Operation not permitted`

for `MAP_ANONYMOUS`  only `failed to map memory: Invalid argument` is displayed

Falcosc, 2021-04-06:
Do you know how to properly maintain the AUR packages?

Maybe you could help @delbato to add the release build flag and version handling into https://aur.archlinux.org/packages/ryzenadj-git/

After we are done with your issue this could also get a requirement for https://aur.archlinux.org/packages/ryzen_smu-dkms-git/

And if you are really cool, you could make it as an optional dependency which is only used if the memory policy doesn't fit.

Abh15h3k, 2021-04-06:
> Do you know how to properly maintain the AUR packages?

eehhh no. i'll look into and see if i can help with this.

Falcosc, 2021-04-06:
I don't even know how this works. As it turns out, any stranger could add packages there. Don't know how this concept is supposed to work if this stranger losses interest into maintaining the configuration, like it did there ;)

Would be nice if you could collect some arch linux knowledge to clean this mess up.

Falcosc, 2021-04-06:
@Abh15h3k I did implement a proof on concept. If it does work, I will add all needed conditions to use this kernel module only if really necessary.

Abh15h3k, 2021-04-06:
Nice!

`--info` and `--dump-table` seem to be working as expected.

POC `sudo ./ryzenadj --info` : https://pastebin.com/fzgBNTpp
POC `sudo ./ryzenadj --dump-table --stapm-limit 30000` : https://pastebin.com/aG1FwUjN

Falcosc, 2021-04-06:
Ok, lets try the real version, `kernelmodule` branch in my repo.
I did create a new branch because I didn't have any patience 😃

Falcosc, 2021-04-06:
If the branch `kernelmodule` does work to, please simulate a non existing ryzen_smu module by changing the name of the driver here:
https://github.com/Falcosc/RyzenAdj/blob/kernelmodule/lib/libpci.c#L60

Abh15h3k, 2021-04-06:
TLDR: working correctly and shows your error message if kernel module is not installed and loaded.

`kernelmodule` branch is also working, shows info and dump-table with correct before and after.

changing `ryzen_smu_drv` to `not_ryzen_smu_drv` gives your error message. ( i also did one with the correct driver name and just unloaded the mode `sudo rmmod ryzen_smu` which also give your error message )

```
failed to map /dev/mem: Operation not permitted
If you don't want to change your memory access policy, you need a kernel module for this task.
We do support usage of this kernel module https://gitlab.com/leogx9r/ryzen_smu
refresh_table failed
Unable to init power metric table: -5
```

Falcosc, 2021-04-06:
Thank you very much. I will prepare the version increment and put everything into the pull request later.
But because we introduce access of a new kernel module, the merge may take some days until @FlyGoat is available to approve the integration of the new kernel module.

Abh15h3k, 2021-04-06:
Great.
No problem with the delay in merging.
Thanks for the quick responses.

You can close issue now or once the changes have been merged.

Falcosc, 2021-04-06:
@Abh15h3k did you found out how to maintain the AUR?

Abh15h3k, 2021-04-06:
Yeah. i have a basic understanding how AUR packages are made and put up in the AUR. just wondering if the current maintainer of `ryzenadj-git` has actually orphaned the package. if so i have to see how to take over as the maintainer.

other than that adding `ryzen_smu-git` as an optional dependency is a one line addition to the PKGBUILD.

Falcosc, 2021-04-06:
Thank you for helping. I will tell you when it is merged.

Abh15h3k, 2021-04-06:
OK. so i've taken over as the maintainer of the AUR package. It's no longer flagged out of date (it shouldn't have been in the first place, since it pulls the git to build the package). I'll bump the pkgver to 0.8.1 (from your commits) once the merge is done,

Falcosc, 2021-04-06:
Technically this is wrong today, because master is  at 0.8.0 as long as this pull request is still open.

But thanks anyway

The only thing I wasn't sure about is the build variant. Cmake will create debug by default. Not sure what AUR does by default.

Abh15h3k, 2021-04-07:
Yeah. The AUR was building the debug version. fixed that and i just bumped the version to the one generated by the pkgver() section of the PKGBUILD (generates a package version using git describe, which changes whenever a commit is push to upstream).

Falcosc, 2021-04-07:
Interesting concept for a package version. I mean if we had a breaking API change you would need to distinguish between the latest version versions maintained in tags.

Is each repo update delivered to everybody even if the AUR version string doesn't change? Or does it only affect new installations?

Abh15h3k, 2021-04-07:
i'm pretty fresh on how this works too. but my guess would be once a user installs the package they won't see the package as updated till i manually change the pkgver. in that situation the user reinstall `yay -S ryzenadj-git` to to get the "latest and greatest" version.

since you brought this up. i was wondering if i should make a non `-git` aur package which will download the latest tagged 
"stable" release so incase there's a commit that breaks something. users can use the non `-git` package which would get updated only when FlyGoat creates a new tagged "stable" release

Falcosc, 2021-04-07:
That would be best-case scenario.

Having versions like 0.8.0 for tags
and versions like 0.8.1.r[githash] for nightly builds.

May you can automate the pkgver for nightly builds to notify all archlinux users about a new merged commit :)

I guess the outdated complaint are from this direct "AUR does not automatically update my installation if nobody changes the version string"

Abh15h3k, 2021-04-08:
@Falcosc what sounds better for the description of the optional dependency
`kernel module needed to access pm_table on systems with strict memory access policy`
or
`kernel module needed to access pm_table on hardened systems`

Falcosc, 2021-04-08:
@Abh15h3k please use the first one because it is all about memory.

Abh15h3k, 2021-04-08:
I'll leave it as a optional dependency.
My reasoning is, without this installed ryzenadj still works as it should. we just can't see the table (correct me if i'm wrong, it had been working for me before i opened the issue).
another thing is that since `ryzen_smu-dkms-git` is another AUR package. `pacman` (apt for arch linux) cannot automatically install it (as it does with other dependencies available in normal repositories.

Falcosc, 2021-04-08:
make sense.

Yes it does work without pmtable

commit c9b51b79 (2021-04-08): Add Support for ryzen_smu kernel module as memory access solution
resolves #133