FlyGoat/RyzenAdj #138: Multithreading and multiple processes could cause issues

Falcosc, 2021-04-12:
How to reproduce it:

- limit CPU to 400Mhz to slow things down
- execute multiple tools which are talking to MP1 with 10 calls / second
  - ryzenAdj via lib with 10 calls / second
  - HWiNFO with 10 calls / second
  - some manually executed ryzenAdj CLI calls

Because we don't check if it is ok to clear the response register and if it is ok to write argument registers, this will lead into problems.
Normally, we don't see them because:
- some commands don't need any arguments (0% chance of issue on argument overwrite)
- some commands only check 1st argument (16.6% chance of issue on argument overwrite)
- we check the response register very fast by doing a loop without additional operations (nearly impossible that something did clear the response register before we did read it and even if something was faster than our code, if the 2nd command does return OK as well, we don't see any problem)

I could find on issue after 1000's of calls: https://github.com/FlyGoat/RyzenAdj/blob/master/lib/nb_smu_ops.c#L51
- argument register was cleared between writing it and reading it
- error messages "PCI Bus is not writeable, check secure boot" was triggered
- this cannot be prevented because we already don't do additional commands between write and read

@FlyGoat what do you think about this? I would say this isn't an issue because it is very unlikely to happen. But I am still not sure if it is good to clear the response register without checking anything else. But I don't know how this low-level stuff should be handled correctly. You will know it much better than me :)

FlyGoat, 2021-04-12:
Just tell users not to do so.
Actually we don't have any possible solution....

Falcosc, 2021-04-12:
@sbski told about a lock on PCI config space. I'm not aware that such an extreme kind of lock is even possible. 1usmus does use this kind of lock.

I would like to know if this kind of lock could hurt something else before we use it to fix something which only very rarely occur (if at all)
I'm still not convinced that that is even a thing ;)