FlyGoat/RyzenAdj #141: Very good softwarre

jumpingjackflash5, 2021-04-15:
Thank your for this program I am using it with Ryzen 3200U.
I use simple scheduling - command line - with reapply after wakeup from S3 and after bootup. Works fine for keeping the parameters set.

Falcosc, 2021-04-15:
I did need to add Power Source Change event to my task scheduler. But I got timing issues at some point and switch to a loop. (stock profile from Power Source change got applied after ryzenAdj and overwrite it)

jumpingjackflash5, 2021-04-15:
Thank you for answer. I use just event after S3 wakeup (and startup) and then run ryzenadj.exe with correct parameters.

[image]

Falcosc, 2021-04-15:
If you want to add power source change, you need to prove a XML configuration for the query if I remember correctly.

I did a quick look, you can also add the power source into the trigger to have 2 actions, one for battery and one for AC

```
    <QueryList> 
      <Query Id="0" Path="System"> 
        <Select Path="System"> 
           *[System[Provider[@Name='Microsoft-Windows-Kernel-Power'] and EventID=105] and EventData[Data[@Name='AcOnline']='false'] ]    
        </Select> 
      </Query> 
    </QueryList> 
```

jumpingjackflash5, 2021-04-15:
Yes it may be done this way. But for me it enough to reapply one set of settings.