FlyGoat/RyzenAdj #143: v0.8 blocked by Riot Vanguard Anticheat

ahalekelly, 2021-04-20:
When I run v0.8 with Vanguard Anticheat running, I get the following error:
```CPU Family: Raven
SMU BIOS Interface Version: 5
Version: v0.8.0
Could not open inpoutx64 driver
Unable to get MEM Obj
Unable to init power metric table: -5
```
v0.8 runs normally if you close Vanguard, as does v0.7.1 even while Vanguard is running.
Vanguard doesn't say anything when you run ryzenadj, but on startup Vanguard says it blocked inpoutx64.sys, even though ryzenadj doesn't run on startup. This just started recently, but I uninstalled all the software that I thought could be triggering it.

Falcosc, 2021-04-21:
Adjustments should still work.

Only your access to monitoring values (--info and --dump-table) will be blocked because RyzenAdj is not part of the Vanguard whitelist.

Falcosc, 2021-04-21:
`inpoutx64.sys` is the driver for direct memory access. Some tools use it, we do use it since v0.8.0 too.
But we made it optionally. So either Vanguard blocks it even nobody does load it or you still have something in startup which does use it or something is calling ryzenAjd --info

@ahalekelly Could you delete the file and tell me if this does help? I will add "delete inpoutx64.sys" as workaround to avoid the blocked message.

ahalekelly, 2021-04-21:
Oh you're right! Adjustments still work, it's just -i that was failing.
Still no idea what's calling the error message. I disabled every startup item and it still happens. I renamed the inpoutx64.sys and the error message went away, but I'd rather not delete a core system driver, I'd rather just live with the error message.

PR #144 (2021-04-27): Improve Memory Access Error Message
Resolves #143
commit c392de18 (2021-04-27): Improve Memory Access Error Message
Resolves #143
closed 2021-04-21 by ahalekelly