FlyGoat/RyzenAdj #145: Cezanne support

prj, 2021-05-05:
I have a 5600U in a HP 455 G8.
It is mostly limited by STAPM, but skin temp shows 0.

Here is an example when running CPU-Z stress test...
[image]

Falcosc, 2021-05-05:
Interesting, you may don't have SSTv2.

Try set --apu-skin-temp to see if your value does appear in this table.

This does also tell why you have STAPM control.

Devices SSTv2 don't have STAPM control but skin temp control.

prj, 2021-05-05:
I tried setting it, it does appear in this table, but I don't see that it does anything.
It behaves like a 3rd timer creeping up, and eventually (slowly) it gets to it.

1. It boosts to PPT limit fast 30W
2. Then the slow limit counts up and it locks to 25W
3. Finally albeit very slowly the STAPM value gets up to 15W and then it locks to 15W, this takes 5 minutes or so.

Falcosc, 2021-05-05:
> I tried setting it, it does appear in this table, but I don't see that it does anything.

Thank you. That tells us that the table is correct.

Yes you don't have SSTv2

if SSTv2 is disable you get STAPM control and SlowPPTTimeConstant control.
But you loose SST Limit control.

I did update the SST dependencies for the STT limits to make this more clear: https://github.com/FlyGoat/RyzenAdj/wiki/Supported-Models

prj, 2021-05-05:
I tried modifying the time constants but they do not change.
STAPM can not be changed either, only the PPT limits can be decreased.

Falcosc, 2021-05-05:
> 
> 
> I tried modifying the time constants but they do not change.
> STAPM can not be changed either, only the PPT limits can be decreased.

You should be able to set 14.000 STAPM, because your device is limited to 15W
https://github.com/FlyGoat/RyzenAdj/wiki/Supported-Models#custom-value-range-for-zen3
The 15.001 does tell that your device is limited at 15.000. If you apply any value above 15.000 it will set 15.001

prj, 2021-05-05:
No, they do not change. That's my entire point.
I can not change absolutely anything.

It says that it set it successfully but there is no difference:
[image]

prj, 2021-05-05:
Could modifying the DSDT help?
Or does this write the same type of ACPI registers as a DSDT mod?

prj, 2021-05-05:
When I set --stapm-time on this, it changes the entry 0x0918 in the dump table.
But RyzenAdj -i does not change.

Falcosc, 2021-05-05:
I am not sure if the DSDT mod does the same thing. But I would like to fix the stamp time readout. Could you please shere your dump table?

If you combine adjustments and --dump-table in the same ryzenAdj call, it will print an additional after adjustment column.

prj, 2021-05-05:
Of course.
```ryzenadj.exe --stapm-time=400 --dump-table > dump.txt```
[test.txt](https://github.com/FlyGoat/RyzenAdj/files/6429493/test.txt)

Falcosc, 2021-05-05:
Thank you, I now have to find out which table versions are affected by this bug.

prj, 2021-05-05:
Also slow-time is at 0x091C on this.

Falcosc, 2021-05-05:
Thank you, I did just delete my text because I was about to ask you this question ;)

prj, 2021-05-05:
Btw, changing these values works fine.
When I reboot they are set to 275 and 5.
I can set them to 500 and 30. I get a much longer duration of boost this way. Thermals are OK.

Falcosc, 2021-05-05:
Unfortunately, setting large time values as workaround has a flaw. If you manage to utilize this limit after 1 hour of lead, it will take 1 hour of idle to restore the limit.

If you have luck, it allows setting ridiculous high values to prevent STAPM limit the whole day and reboot does reset the utilization.

prj, 2021-05-05:
The highest it goes is 500 and 30.
This would allow 8 minutes and 20 seconds of 25W boost. I think that's good for many use cases.
Not good if you want to do some long term processing...

When I get some time I could try the DSDT method. Someone did write that when RyzenAdj was unable to change limit, with the DSDT mod it worked afterwards. But this was not about Cezanne.
No idea where the limit is set, but I'd wager a guess, that it should be somewhere in the bios.

Falcosc, 2021-05-05:
> Someone did write that when RyzenAdj was unable to change limit, with the DSDT mod it worked afterwards. But this was not about Cezanne.

In the time when this mod was discovered we didn't know about period value overwrites. So we don't know if DSDT mod does only fix issues with value overwrite or if it can resolve the Zen3 cap. The Zen3 cap is a new thing, we didn't see this behavior on any other generation.

But I wish you luck, you will be the first one who is having a limit issue and willing to share the result.

I will add your result in our wiki.
Thank you

prj, 2021-05-07:
This seems to display the time correctly, however I can not get the readjust service to work.
What is the correct value to put into ```$monitorField``` for stapm time?

prj, 2021-05-08:
A comment about stapm-time.
If you set stapm-time to zero it instantly changes stapm-value to the current utilization.

So, as a workaround, it is possible to monitor current usage (PPT-FAST? Or maybe there is another entry in the table that shows instantaneous TDP), then do something like this:
```$stapm_time_target = 500
$stapm_hysteresis = 1

if (current_value < (stapm_value - $stapm_hysteresis)) {
	set_stapm_time(0);
	# Maybe tiny sleep needed here
	set_stapm_time($stapm_time_target);
}```

This will reset the stapm instantly...
I will test the dsdt bios method and see what that does, if it fails, I will probably implement this hack in the reAdjustService.ps1 script.

Falcosc, 2021-05-08:
Nice trick.
Do you need `set_stapm_limit(10)` for this trick?
Don't forget the refresh_table call at the beginning of your tick because the pm table getter access the last stored table result.

We can add it to the wiki after your conclusion.

prj, 2021-05-08:
> 
> 
> Nice trick.
> Do you need `set_stapm_limit(10)` for this trick?
> Don't forget the refresh_table call at the beginning of your tick because the pm table getter access the last stored table result.
> 
> We can add it to the wiki after your conclusion.

Yes, you do.
Let's say you have some long task (encoding etc) and are getting close to 14 or 14.5 on the `stapm_value` and your `stapm_limit` is 15.
Now would be good time to reset it, because it will start throttling as soon as it hits 15 to try to keep it there.
But if you simply set the time to 0, `stapm_value` will just jump to 25 immediately, because that's what it running at that moment.
If you additionally set `stapm_limit` to 10, then it will throttle from 25->10.
Next moment you set time to 0, it picks up the 10, to which it throttled (will need to do some testing how long the delay must be), now your `stapm_value` is at 10. Then you set the time back to maximum (500 in my case) and restore `stapm_limit` to 15.

End result: `stapm_value` is at 10, `stapm_limit` at 15, and `stapm_time` is 500 again, so it goes to the 25W boost for a little over 2 minutes, at which point `stapm_value` creeps back up to 14.5 or so, and you just repeat the process again.

So you can have a tiny interruption on the CPU when you're throttling it down, but I think for a long running encoding or processing task realistically you will be running at 25W all the time (apart from the <500ms every 2 minutes).

Falcosc, 2021-05-08:
your while loop doesn't work

>  Don't forget the refresh_table call because the pm table getter access the last stored table result.

prj, 2021-05-08:
It is just pseudocode for now :)

Falcosc, 2021-05-08:
You can use much lower delays, CPU Power management isn't that slow ;)

prj, 2021-05-09:
I realized that powershell Sleep is in seconds, not in milliseconds.
This approach works hilariously well though.

I ran the CPU-Z stresser and tested.
Without the hack, I have an all core freq of 2.6-2.7ghz after the `stapm_value` hits the `stapm_limit`.
With the hack I have an all core freq of 3.4ghz, which dips for a very short moment to 3.1ghz on reset.
It seems that the thermal management kicks in about 0.9W before the `stapm_limit`, so a hysteresis of 1 works very well.

Here is an example output with `stapm_time` of 100. I cut out some of the lines, so it is not overly long. Pretty funny loophole they left in.
``
C:\Program Files\RyzenAdj>powershell ./readjustService.ps1
Test Adjustments
set new monitoring target stapm_time to 100
set new monitoring target stapm_time to 100
Monitor stapm_time changes and PowerSlider changes every 1 seconds...
[STAPM_RESET] Limit: 15 Value: 7.92096
...
[STAPM_RESET] Limit: 15 Value: 13.91258
[STAPM_RESET] Limit: 15 Value: 14.0097
[STAPM_RESET] stapm_value (14.0097) nearing stapm_limit (15), resetting...
[STAPM_RESET] iter: 0 stapm_value: 13.91153
[STAPM_RESET] iter: 1 stapm_value: 10.15482
[STAPM_RESET] Reset successful, stapm_value now at 10.15482, needed 1 iterations.
[STAPM_RESET] Limit: 15 Value: 10.63961
[STAPM_RESET] Limit: 15 Value: 10.87728
...
[STAPM_RESET] Limit: 15 Value: 13.90829
[STAPM_RESET] Limit: 15 Value: 14.02029
[STAPM_RESET] stapm_value (14.02029) nearing stapm_limit (15), resetting...
[STAPM_RESET] iter: 0 stapm_value: 15.74085
[STAPM_RESET] iter: 1 stapm_value: 10.20906
[STAPM_RESET] Reset successful, stapm_value now at 10.20906, needed 1 iterations.
[STAPM_RESET] Limit: 15 Value: 10.23359
[STAPM_RESET] Limit: 15 Value: 10.38286
...
[STAPM_RESET] Limit: 15 Value: 13.99841
[STAPM_RESET] Limit: 15 Value: 14.11627
[STAPM_RESET] stapm_value (14.11627) nearing stapm_limit (15), resetting...
[STAPM_RESET] iter: 0 stapm_value: 16.08081
[STAPM_RESET] iter: 1 stapm_value: 10.3063
[STAPM_RESET] Reset successful, stapm_value now at 10.3063, needed 1 iterations.
[STAPM_RESET] Limit: 15 Value: 10.3679
[STAPM_RESET] Limit: 15 Value: 10.51569
```
Here is the code I added, this is the first time I have seen PowerShell, so please do not be overly critical :)
Top of file:
```
$stapm_time_target = 100
$stapm_normal_limit = 15000
$stapm_reset_limit = 10000
$stapm_hysteresis = 1
```

Bottom of file:
```
if($monitorField -or $updateHWINFOSensors) {
	[void][ryzen.adj]::refresh_table($ry)
	#[System.Runtime.InteropServices.Marshal]::Copy($tablePtr, $pmTable, 0, 560);
	
	# Workaround for locked stapm_time
	$cur_limit = [ryzen.adj]::get_stapm_limit($ry)
	$cur_value = [ryzen.adj]::get_stapm_value($ry)
	$iters = 0
	
	Write-Host "[STAPM_RESET] Limit: $cur_limit Value: $cur_value"
	
	if ($cur_value -gt ($cur_limit - $stapm_hysteresis)) {
		Write-Host "[STAPM_RESET] stapm_value ($cur_value) nearing stapm_limit ($cur_limit), resetting..."
		$result = [ryzen.adj]::set_stapm_limit($ry, $stapm_reset_limit)
		$result = [ryzen.adj]::set_stapm_time($ry, 0)
		while ($true) {
			Start-Sleep -Milliseconds 10
			[void][ryzen.adj]::refresh_table($ry)
			$cur_limit = [ryzen.adj]::get_stapm_limit($ry)
			$cur_value = [ryzen.adj]::get_stapm_value($ry)
			Write-Host "[STAPM_RESET] iter: $iters stapm_value: $cur_value"
			if ($cur_limit -ne ($stapm_reset_limit / 1000)) {
				Write-Host "[STAPM_RESET] Reset failed, something else changed stapm_limit."
				break
			}
			if ($cur_value -lt (($stapm_reset_limit / 1000) + $stapm_hysteresis)) {
				Write-Host "[STAPM_RESET] Reset successful, stapm_value now at $cur_value, needed $iters iterations."
				break
			}
			$iters++
		}
		$result = [ryzen.adj]::set_stapm_time($ry, $stapm_time_target)
		$result = [ryzen.adj]::set_stapm_limit($ry, $stapm_normal_limit)
		[void][ryzen.adj]::refresh_table($ry)
	}
}
```
Goodbye 15W TDP limit...

Falcosc, 2021-05-10:
I did add it directly to the powershell script because it is possible to configer the whole thing just with a simple `$resetSTAPMUsage` flag

https://github.com/FlyGoat/RyzenAdj/pull/148/files

prj, 2021-05-10:
The trouble with not checking if it is what you set it to, is a potential infinite loop. I see you did 3 iteration limit. I would do more like 10 iteration limit.
I did not know about the on-battery stuff, I am going to check it later today on my laptop and see if that is the case.

Falcosc, 2021-05-10:
Have a look on your max iteration count. I did change the sleep to 20ms, because you never got it working with 10ms

If it is not done in 3-times the normal reset duration, then something is broken and the old limit and old time should be restored

prj, 2021-05-10:
Actually what works first time is around 200ms.
My theory is that on the second iteration around that time passes due to the function calls to the DLL...

If you want I can add some more logging and measure actual elapsed time.

FlyGoat, 2021-05-10:
Just a kindly reminder, the smu is designed to handle pulls several times per second, not hundereds times. It may have some erroneous behavior at this frequency. 20ms isn't really reasonable.

prj, 2021-05-10:
I agree, I think it isn't necessary either. I'll make some measurements and see what the realistic times are.

Falcosc, 2021-05-12:
@prj please close HWInfo and other tools which do use PM Table to measure the needed time because we reuse the PM Table result to avoid transfer table race conditions.

Example order of calls:
0ms: refresh_table (get a new table)
10ms: noop
20ms: noop (HWiNFO did request a new table)
30ms: noop (limit reset is finished)
100ms: refresh_table (reuses the result from HW info and tells that limit reset is not done because result is from T+20ms)
200ms: refresh_table (get a new table)

So test result would tell that you needed to wait 200ms, but actually the fix was done after 30ms

prj, 2021-05-12:
Nothing else was open.
Will do tests today/tomorrow. I swapped to a 445 G8, which is the same PCB's just in a smaller chassis.

prj, 2021-05-13:
Okay, so I did some tests.
As I thought, it's not so simple.

With the code as I posted it and 10ms loop, it always fails on first iteration and succeeds on second. The time taken is 45-65ms.
If I add a 50ms delay before starting the loop (so total of 60ms), and then retry every 10ms, the result is 73-128ms. Sometimes it needs 1 iteration, sometimes 2.
With 90ms + 10ms delay (total 100ms), then retrying every 10ms the result is 120-150ms, and sometimes it still needs 1 extra iteration.

I think the problem is that sleep actually sleeps for different times, and with heavy CPU load it is super imprecise.
In a tight loop without any delay it takes 8-36ms.

The best compromise is a 1ms sleep, so it yields the thread between calls and what also helps is higher process priority than "Normal" or reducing the stresser to "Low".

With "High" priority and 1ms it always does 2 iterations, and always manages exactly 30ms. With normal priority it takes between 30 and 60ms.
If someone is gaming for example, then if we throttle it down too long it will cause microstutter. So I think a good way to do it is Sleep(1) and then 10 iteration limit.
I don't see any artifacts or issues with the table, unless I run multiple tools at the same time and run it in a tight loop...

prj, 2021-05-13:
Example with 1ms sleep and "High" priority, I set time to 40 for testing:
```[STAPM_RESET] Limit: 15 Value: 14.09409
[STAPM_RESET] stapm_value (14.09409) nearing stapm_limit (15), resetting...
[STAPM_RESET] iter: 0 stapm_value: 15.45263
[STAPM_RESET] iter: 1 stapm_value: 10.99589
[STAPM_RESET] Reset successful, stapm_value now at 10.99589, needed 1 iterations. Took 24 ms
[STAPM_RESET] Limit: 15 Value: 10.61365
[STAPM_RESET] Limit: 15 Value: 10.97124
[STAPM_RESET] Limit: 15 Value: 11.31946
[STAPM_RESET] Limit: 15 Value: 11.6607
[STAPM_RESET] Limit: 15 Value: 11.99283
[STAPM_RESET] Limit: 15 Value: 12.31519
[STAPM_RESET] Limit: 15 Value: 12.62992
[STAPM_RESET] Limit: 15 Value: 12.93663
[STAPM_RESET] Limit: 15 Value: 13.23724
[STAPM_RESET] Limit: 15 Value: 13.52993
[STAPM_RESET] Limit: 15 Value: 13.81431
[STAPM_RESET] Limit: 15 Value: 14.09217
[STAPM_RESET] stapm_value (14.09217) nearing stapm_limit (15), resetting...
[STAPM_RESET] iter: 0 stapm_value: 15.22284
[STAPM_RESET] iter: 1 stapm_value: 10.22146
[STAPM_RESET] Reset successful, stapm_value now at 10.22146, needed 1 iterations. Took 31 ms
[STAPM_RESET] Limit: 15 Value: 10.66883
[STAPM_RESET] Limit: 15 Value: 11.026
[STAPM_RESET] Limit: 15 Value: 11.37224
[STAPM_RESET] Limit: 15 Value: 11.70953
[STAPM_RESET] Limit: 15 Value: 12.03826
[STAPM_RESET] Limit: 15 Value: 12.3587
[STAPM_RESET] Limit: 15 Value: 12.67139
[STAPM_RESET] Limit: 15 Value: 12.97618
[STAPM_RESET] Limit: 15 Value: 13.27355
[STAPM_RESET] Limit: 15 Value: 13.56359
[STAPM_RESET] Limit: 15 Value: 13.84859
[STAPM_RESET] Limit: 15 Value: 14.12682
[STAPM_RESET] stapm_value (14.12682) nearing stapm_limit (15), resetting...
[STAPM_RESET] iter: 0 stapm_value: 15.38339
[STAPM_RESET] iter: 1 stapm_value: 10.13629
[STAPM_RESET] Reset successful, stapm_value now at 10.13629, needed 1 iterations. Took 31 ms
[STAPM_RESET] Limit: 15 Value: 10.7365
[STAPM_RESET] Limit: 15 Value: 11.08908
[STAPM_RESET] Limit: 15 Value: 11.43318
[STAPM_RESET] Limit: 15 Value: 11.77113
[STAPM_RESET] Limit: 15 Value: 12.10034
[STAPM_RESET] Limit: 15 Value: 12.41963
[STAPM_RESET] Limit: 15 Value: 12.73149
[STAPM_RESET] Limit: 15 Value: 13.03724
[STAPM_RESET] Limit: 15 Value: 13.33503
[STAPM_RESET] Limit: 15 Value: 13.62367
[STAPM_RESET] Limit: 15 Value: 13.90686
[STAPM_RESET] Limit: 15 Value: 14.18318
[STAPM_RESET] stapm_value (14.18318) nearing stapm_limit (15), resetting...
[STAPM_RESET] iter: 0 stapm_value: 12.26344
[STAPM_RESET] iter: 1 stapm_value: 10.07343
[STAPM_RESET] Reset successful, stapm_value now at 10.07343, needed 1 iterations. Took 28 ms
[STAPM_RESET] Limit: 15 Value: 10.27712
[STAPM_RESET] Limit: 15 Value: 10.6435
[STAPM_RESET] Limit: 15 Value: 11.00018
[STAPM_RESET] Limit: 15 Value: 11.35012
[STAPM_RESET] Limit: 15 Value: 11.69004
[STAPM_RESET] Limit: 15 Value: 12.02189
[STAPM_RESET] Limit: 15 Value: 12.34518
[STAPM_RESET] Limit: 15 Value: 12.66049
[STAPM_RESET] Limit: 15 Value: 12.96814
[STAPM_RESET] Limit: 15 Value: 13.26672
[STAPM_RESET] Limit: 15 Value: 13.55957
[STAPM_RESET] Limit: 15 Value: 13.8447
[STAPM_RESET] Limit: 15 Value: 14.1207
[STAPM_RESET] stapm_value (14.1207) nearing stapm_limit (15), resetting...
[STAPM_RESET] iter: 0 stapm_value: 15.84294
[STAPM_RESET] iter: 1 stapm_value: 10.13397
[STAPM_RESET] Reset successful, stapm_value now at 10.13397, needed 1 iterations. Took 30 ms
[STAPM_RESET] Limit: 15 Value: 11.2886
[STAPM_RESET] Limit: 15 Value: 11.62971
[STAPM_RESET] Limit: 15 Value: 11.96024
[STAPM_RESET] Limit: 15 Value: 12.28276
[STAPM_RESET] Limit: 15 Value: 12.59767
[STAPM_RESET] Limit: 15 Value: 12.90547
[STAPM_RESET] Limit: 15 Value: 13.20544
[STAPM_RESET] Limit: 15 Value: 13.49752
[STAPM_RESET] Limit: 15 Value: 13.78254
[STAPM_RESET] Limit: 15 Value: 14.06091
[STAPM_RESET] stapm_value (14.06091) nearing stapm_limit (15), resetting...
[STAPM_RESET] iter: 0 stapm_value: 16.2242
[STAPM_RESET] iter: 1 stapm_value: 10.19323
[STAPM_RESET] Reset successful, stapm_value now at 10.19323, needed 1 iterations. Took 30 ms
[STAPM_RESET] Limit: 15 Value: 10.90912
[STAPM_RESET] Limit: 15 Value: 11.25765
[STAPM_RESET] Limit: 15 Value: 11.59763
[STAPM_RESET] Limit: 15 Value: 11.93103
[STAPM_RESET] Limit: 15 Value: 12.25825
[STAPM_RESET] Limit: 15 Value: 12.57528
[STAPM_RESET] Limit: 15 Value: 12.88247
[STAPM_RESET] Limit: 15 Value: 13.18296
[STAPM_RESET] Limit: 15 Value: 13.47614
[STAPM_RESET] Limit: 15 Value: 13.76156
[STAPM_RESET] Limit: 15 Value: 14.04149
[STAPM_RESET] stapm_value (14.04149) nearing stapm_limit (15), resetting...
[STAPM_RESET] iter: 0 stapm_value: 16.43928
[STAPM_RESET] iter: 1 stapm_value: 10.14261
[STAPM_RESET] Reset successful, stapm_value now at 10.14261, needed 1 iterations. Took 30 ms
[STAPM_RESET] Limit: 15 Value: 10.74394
[STAPM_RESET] Limit: 15 Value: 11.09816
[STAPM_RESET] Limit: 15 Value: 11.44405
[STAPM_RESET] Limit: 15 Value: 11.78097
[STAPM_RESET] Limit: 15 Value: 12.10802
[STAPM_RESET] Limit: 15 Value: 12.42766
[STAPM_RESET] Limit: 15 Value: 12.73997
[STAPM_RESET] Limit: 15 Value: 13.04475
[STAPM_RESET] Limit: 15 Value: 13.34292
[STAPM_RESET] Limit: 15 Value: 13.63358
[STAPM_RESET] Limit: 15 Value: 13.91657
[STAPM_RESET] Limit: 15 Value: 14.19398
[STAPM_RESET] stapm_value (14.19398) nearing stapm_limit (15), resetting...
[STAPM_RESET] iter: 0 stapm_value: 12.34594
[STAPM_RESET] iter: 1 stapm_value: 10.06669
[STAPM_RESET] Reset successful, stapm_value now at 10.06669, needed 1 iterations. Took 30 ms
[STAPM_RESET] Limit: 15 Value: 10.58164```

Falcosc, 2021-05-13:
So, your 200ms result was a mistake?

Is 20ms waiting a good value to success mostly on first try? Or should we use 30m?

I would like to share https://github.com/FlyGoat/RyzenAdj/pull/148/files with the next release.
Just need your suggestion based on your results.

Falcosc, 2021-05-13:
`[Threading.Thread]::Sleep(10)`
works better then
`Start-Sleep -Milliseconds 10`

If it makes trouble we can also use a timer instead of sleep. Then the waiting times will be much more stable.

prj, 2021-05-13:
I think the problem with timers is you have to configure them and it's system-wide.
Let me try calling it through the CLR.

Sorry, I don't have a lot of experience with PowerShell, but I am pretty good with programming in general.

Falcosc, 2021-05-13:
I tend to avoid the powershell cmdlets, because they are slow ;)

If [Threading.Thread]::Sleep(10) doesn't help, I would at least use a rising sleep solution to avoid use cases where you spam hundreds of calls in a second.

`foreach ($i in @(1,10,100))`

A proper timer is easy to implement, but it looks ugly and is a lot of code because you need to put needed stuff into the timer context, it doesn't have access to the script context. This is unlike anonymous function bodies.
So if a sleep loop does work, I would prefer it.

prj, 2021-05-13:
It is a bit faster with `[Threading.Thread]::Sleep(1)`
I get 23ms-60ms for the whole if...

Here is code I used now:
```
		if ($cur_value -gt ($cur_limit - $stapm_hysteresis)) {
			Write-Host "[STAPM_RESET] stapm_value ($cur_value) nearing stapm_limit ($cur_limit), resetting..."
		    $stopwatch.start()
			$result = [ryzen.adj]::set_stapm_limit($ry, $stapm_reset_limit)
			$result = [ryzen.adj]::set_stapm_time($ry, 0)
			
			while ($iters -lt 10) {
				[Threading.Thread]::Sleep(70)
				[void][ryzen.adj]::refresh_table($ry)
				$cur_limit = [ryzen.adj]::get_stapm_limit($ry)
				$cur_value = [ryzen.adj]::get_stapm_value($ry)
				Write-Host "[STAPM_RESET] iter: $iters stapm_value: $cur_value"
				if ($cur_value -lt (($stapm_reset_limit / 1000) + $stapm_hysteresis)) {
					$elapsed = $stopwatch.ElapsedMilliseconds
					Write-Host "[STAPM_RESET] Reset successful, stapm_value now at $cur_value, needed $iters iterations. Took $elapsed ms"
					break
				}
				$iters++
			}
			$stopwatch.reset()
			
			$result = [ryzen.adj]::set_stapm_time($ry, $stapm_time_target)
			$result = [ryzen.adj]::set_stapm_limit($ry, $stapm_normal_limit)
			[void][ryzen.adj]::refresh_table($ry)
		}
```
For a "closed loop mode" I think this works well. If I totally remove the sleep I just get sometimes 8 calls in a row... And gain maybe 5-10ms.

However, I think maybe I made it all unnecessarily complicated. The tables just don't update in time.
Here is example totally without delay:
```
[STAPM_RESET] Limit: 15 Value: 13.52855
[STAPM_RESET] Limit: 15 Value: 13.72989
[STAPM_RESET] Limit: 15 Value: 14.10654
[STAPM_RESET] stapm_value (14.10654) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 9.90922
[STAPM_RESET] Limit: 15 Value: 10.40666
[STAPM_RESET] Limit: 15 Value: 10.88951
[STAPM_RESET] Limit: 15 Value: 11.35694
[STAPM_RESET] Limit: 15 Value: 11.80646
[STAPM_RESET] Limit: 15 Value: 12.24051
[STAPM_RESET] Limit: 15 Value: 12.66032
[STAPM_RESET] Limit: 15 Value: 13.06906
[STAPM_RESET] Limit: 15 Value: 13.46697
[STAPM_RESET] Limit: 15 Value: 13.85163
[STAPM_RESET] Limit: 15 Value: 14.22009
[STAPM_RESET] stapm_value (14.22009) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 14.99389
[STAPM_RESET] stapm_value (14.99389) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 14.72575
[STAPM_RESET] stapm_value (14.72575) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 14.99474
[STAPM_RESET] stapm_value (14.99474) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 14.9927
[STAPM_RESET] stapm_value (14.9927) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 10.33404
[STAPM_RESET] Limit: 15 Value: 10.98435
[STAPM_RESET] Limit: 15 Value: 11.61806
[STAPM_RESET] Limit: 15 Value: 12.20967
[STAPM_RESET] Limit: 15 Value: 12.65853
[STAPM_RESET] Limit: 15 Value: 13.06577
[STAPM_RESET] Limit: 15 Value: 13.45887
[STAPM_RESET] Limit: 15 Value: 13.83853
[STAPM_RESET] Limit: 15 Value: 14.20526
[STAPM_RESET] stapm_value (14.20526) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 14.99383
[STAPM_RESET] stapm_value (14.99383) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 10.33641
[STAPM_RESET] Limit: 15 Value: 10.97581
[STAPM_RESET] Limit: 15 Value: 11.47657
[STAPM_RESET] Limit: 15 Value: 11.92735
[STAPM_RESET] Limit: 15 Value: 12.36176
[STAPM_RESET] Limit: 15 Value: 12.78234
[STAPM_RESET] Limit: 15 Value: 13.18632
[STAPM_RESET] Limit: 15 Value: 13.57589
[STAPM_RESET] Limit: 15 Value: 13.95615
[STAPM_RESET] Limit: 15 Value: 14.30743
[STAPM_RESET] stapm_value (14.30743) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 14.99356
[STAPM_RESET] stapm_value (14.99356) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 14.81427
[STAPM_RESET] stapm_value (14.81427) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 14.99288
[STAPM_RESET] stapm_value (14.99288) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 14.53149
[STAPM_RESET] stapm_value (14.53149) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 14.99381
[STAPM_RESET] stapm_value (14.99381) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 14.99496
[STAPM_RESET] stapm_value (14.99496) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 14.99431
[STAPM_RESET] stapm_value (14.99431) nearing stapm_limit (15), resetting...
[STAPM_RESET] Limit: 15 Value: 10.33406
[STAPM_RESET] Limit: 15 Value: 10.98469
[STAPM_RESET] Limit: 15 Value: 11.61412
```

I will find the optimal sleep time now. Then no loop needed.

prj, 2021-05-13:
5ms was a little too low sometimes, but 10ms is perfect.

Here is the code, easy:
```
		if ($cur_value -gt ($cur_limit - $stapm_hysteresis)) {
			Write-Host "[STAPM_RESET] stapm_value ($cur_value) nearing stapm_limit ($cur_limit), resetting..."
		    $stopwatch.start()

			$result = [ryzen.adj]::set_stapm_limit($ry, $stapm_reset_limit)
			$result = [ryzen.adj]::set_stapm_time($ry, 0)

			[Threading.Thread]::Sleep(10)

			$result = [ryzen.adj]::set_stapm_time($ry, $stapm_time_target)
			$result = [ryzen.adj]::set_stapm_limit($ry, $stapm_normal_limit)

			$elapsed = $stopwatch.ElapsedMilliseconds
			Write-Host "[STAPM_RESET] Reset successful, stapm_value now at $cur_value, took $elapsed ms"
			$stopwatch.reset()
			
			[void][ryzen.adj]::refresh_table($ry)
		}
```

No need to complicate anything :)

Falcosc, 2021-05-13:
Could it be that the workaround does work even if you don't wait for the PM table changes?

I will change the pullrequest and remove the lopp. I would like it if you could test it because it doesn't need configuration (old values get extracted from PM table before resetting) so less documentation needed :)

Falcosc, 2021-05-13:
@prj I did put in your suggestions

https://github.com/FlyGoat/RyzenAdj/pull/148/files

Please let me know if this reduced version without configuration does still work or if I made a mistake somewhere with the value.

jackloomen, 2023-05-15:
This is what I have been using for months on Linux in case anyone needs it:
My machine is an HP 5600U model.
Need to run as root.
Will loop forever.

```
#!/bin/bash
set -euo pipefail

ryzenadj --stapm-limit 30000 > /dev/null
ryzenadj --slow-limit 30000 > /dev/null
ryzenadj --fast-limit 30000 > /dev/null

while true; do
	INFO=$(ryzenadj -i)
	STAPM_VALUE=$(echo "$INFO" | rg "STAPM VALUE" | grep -Eo '[[:digit:]]+([.][[:digit:]]+)?')
	PPT_FAST_VALUE=$(echo "$INFO" | rg "PPT VALUE FAST" | grep -Eo '[[:digit:]]+([.][[:digit:]]+)?')
	PPT_SLOW_VALUE=$(echo "$INFO" | rg "PPT VALUE SLOW" | grep -Eo '[[:digit:]]+([.][[:digit:]]+)?')
	TEMP=$(echo "$INFO" | rg "THM VALUE CORE" | grep -Eo '[[:digit:]]+([.][[:digit:]]+)?')

	printf "$PPT_FAST_VALUE \t $PPT_SLOW_VALUE \t $STAPM_VALUE \t $TEMP C\n"

	if (( $(echo "$STAPM_VALUE > 14.4" | bc -l) )); then
		echo "STAPM : $(date '+%Y%M%d | %H:%M') --- $STAPM_VALUE"
		ryzenadj --stapm-limit 10000 > /dev/null
		ryzenadj --stapm-time 0 > /dev/null
		sleep 0.02
		ryzenadj --stapm-time 500 > /dev/null
		ryzenadj --stapm-limit 15000 > /dev/null
		sleep 0.02
	fi
	if (( $(echo "$PPT_SLOW_VALUE > 24.4" | bc -l) )); then
		echo "PPT_SLOW : $(date '+%Y%M%d | %H:%M') --- $PPT_SLOW_VALUE"

		ryzenadj --slow-limit 15000 > /dev/null
		ryzenadj --slow-time 0 > /dev/null
		sleep 0.02
		ryzenadj --slow-time 30 > /dev/null
		ryzenadj --slow-limit 25000 > /dev/null
		sleep 0.02
	fi
	sleep 1
done
```

Falcosc, 2023-05-15:
The poor CPU doing all the initialization steps of the ryzenadj application thousands of times. That hurts so much.

But thanks for sharing. If it works, it is good enough, no matter how many CPU cycles are wasted and how many SMU command conflicts are generated with parallel running applications.

By the way, we have an example to show how to script ryzenAdj
Please avoid CLI result parsing, command line user output should not be treated as interface. Use this instead:
https://github.com/FlyGoat/RyzenAdj/blob/master/examples/readjust.py

JCBuck, 2023-07-24:
Hi all,
I've implemented a win32 (some windows console apis) C version of this and added a configurable interval value reset since on thinkpads it is required to bypass stapm and hit fast it seems... Currently defaults to sleeping 12ms a few times per loop.

https://github.com/JCBuck/RyzenAdjTimer

angelokimhui, 2024-05-30:
Hi, another way to bypass it is to adjust cpu_skin_temp to 60.

voidastro4, 2024-07-18:
> Hi, another way to bypass it is to adjust cpu_skin_temp to 60.

Neither --apu-skin-temp=60 nor --dgpu-skin-temp=60 (together) worked.

voidastro4, 2024-07-18:
> The poor CPU doing all the initialization steps of the ryzenadj application thousands of times. That hurts so much.
>
> But thanks for sharing. If it works, it is good enough, no matter how many CPU [...]

This works the same way to inefficient bash script did:

```
adjust("stapm_limit", 30000)
adjust("slow_limit", 30000)
adjust("fast_limit", 30000)

while True:
    lib.refresh_table(ry)
    stapm = lib.get_stapm_value(ry)
    ppt_fast = lib.get_fast_value(ry)
    ppt_slow = lib.get_slow_value(ry)
    print(str(round(ppt_slow, 2)) + "\t" + str(round(stapm, 2)))
    if stapm > 14.2: # decline in clock speed begins around this number
        print('STAPM: ' + str(round(stapm, 2)))
        adjust("stapm_limit", 10000)
        adjust("stapm_time", 0)
        time.sleep(0.01)
        adjust("stapm_time", 500)
        adjust("stapm_limit", 15000)
        time.sleep(0.01)
    if ppt_slow > 24.2:
        print('PPT_SLOW: ' + str(round(ppt_slow, 2)))
        adjust("slow_limit", 15000)
        adjust("slow_time", 0)
        time.sleep(0.01)
        adjust("slow_time", 30)
        adjust("slow_limit", 25000)
        time.sleep(0.01)
    time.sleep(1)
```

Any suggestions to improve this?

Falcosc, 2024-07-18:
no need for improvement. The only major topic was avoiding initalisation for every little call. The old version did initialize up to 5 times every second. Now you only init once per startup :)

Any additional improvements are not worth your time.